arkhipenko/TaskScheduler #4: Compiling Error

mirko314, 2015-10-21:
Hello,
I'm having problems running your example Code. I'm getting the following compiling error.

```
in file included from sketch_oct21b.ino:1:0:
/home/mirko/Arduino/libraries/TaskScheduler/src/TaskScheduler.h:198:1: error: prototype for 'Task::Task(long unsigned int, long int, void (*)(), Scheduler*, bool, bool (*)(), void (*)())' does not match any in class 'Task'
 Task::Task( unsigned long aInterval, long aIterations, void (*aCallback)(), Scheduler* aScheduler, bool aEnable, bool (*aOnEnable)(), void (*aOnDisable)() ) {
 ^
/home/mirko/Arduino/libraries/TaskScheduler/src/TaskScheduler.h:133:7: error: candidates are: constexpr Task::Task(Task&&)
 class Task {
       ^
/home/mirko/Arduino/libraries/TaskScheduler/src/TaskScheduler.h:133:7: error:                 constexpr Task::Task(const Task&)
/home/mirko/Arduino/libraries/TaskScheduler/src/TaskScheduler.h:136:2: error:                 Task::Task(long unsigned int, long int, void (*)(), Scheduler*, boolean, bool (*)(), void (*)())
  Task(unsigned long aInterval=0, long aIterations=0, void (*aCallback)()=NULL, Scheduler* aScheduler=NULL, boolean aEnable=false, bool (*aOnEnable)()=NULL, void (*aOnDisable)()=NULL);
```

arkhipenko, 2015-10-22:
Hi,
I just recompiled all examples on my machine and didn't get any errors.

Could you please specify:

1. What version of the TaskScheduler library you are using?
2. What version of the Arduino IDE you are using? Which platform?
3. What is the target microcontroller?
4. Which of the examples you are compiling? While line generated the error?

For instance, in my case:
1,  v 1.8,0
2. 1.6.5/Windows
3. Uno
4. All of them

Thanks.

Markolise, 2015-10-22:
I am also having a compile error very similar on the older IDE I've been using, Just curious if you've tested this with 1.0.5.
I'm downloading 1.6.6 to use your great library though, so if it doesn't work no big deal for me.

arkhipenko, 2015-10-23:
Have not tested this on 1.0.5.

1.6.5. Should work fine.

https://github.com/arkhipenko/TaskScheduler/issues/4#issuecomment-150382854

closed 2015-10-23 by arkhipenko