arkhipenko/TaskScheduler #5: Warning, Category

Markolise, 2015-11-03:
Not really a big issue, but getting this warning "WARNING: Category 'Schedulers' in Library TaskScheduler is not valid. Setting to 'Uncategorized'"
Any idea how to fix that, or is there something I need to modify in the arduino settings to make that category valid?

arkhipenko, 2015-11-03:
There is no appropriate category for TaskScheduler in the Arduino IDE library manager. I will ask the developers. Please ignore this warning, it does not affect functionality whatsoever.

https://github.com/arkhipenko/TaskScheduler/issues/5

arkhipenko, 2015-11-03:
Below is a list of library categories currently supported by IDE:

Display
Communication
Signal Input/Output
Sensors
Device Control
Timing
Data Storage
Data Processing
Other
Uncategorized

don't see TaskScheduler fitting into any of them cleanly. Perhaps I should make it "Timing", or "Device Control", or "Other"?

Hannes-III, 2015-11-03:
I vote for Timing! That's how I found your Library in the first place.

arkhipenko, 2015-11-03:
Timing it is then. Updated library.properties file.

closed 2015-11-03 by arkhipenko