arkhipenko/TaskScheduler #7: Request: pass in the task and the scheduler to the callback function

patmoore, 2015-11-15:
this will allow a callback to be used for multiple tasks.

arkhipenko, 2015-11-15:
Please review the documentation. It is already done.

patmoore, 2015-11-15:
What documentation?

But more importantly

On line 627:

```
                                if ( iCurrent->iCallback ) {
                                        ( *(iCurrent->iCallback) )();
```
this should be:

```
                                if ( iCurrent->iCallback ) {
                                        ( *(iCurrent->iCallback) )(iCurrent);
```

arkhipenko, 2015-11-16:
O come on, really? The documentation in the extras folder:

https://github.com/arkhipenko/TaskScheduler/tree/master/extras

The Scheduler class already provides access to currently executing task. 
    inline Task& Scheduler::currentTask() {return *iCurrent; }

Most of the tasks do not need this so I decided not to waste time pushing the reference to stack and back all the time.

Refer to example 7 for usage:

https://github.com/arkhipenko/TaskScheduler/blob/master/examples/Scheduler_example7_WDT/Scheduler_example7_WDT.ino

closed 2015-11-16 by arkhipenko

patmoore, 2015-11-16:
That location is not a normal place for documentation. 

Maybe you could create the gh-pages branch? see: https://help.github.com/articles/creating-project-pages-manually/

arkhipenko, 2015-11-16:
For Arduino library they suggest documentation is placed in the extras folder:
\
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata