arkhipenko/TaskScheduler #8: HELP: Flash for chip ID 7e06e0e6 is not supported

jjol7166, 2015-11-17:
Hey mate, 

Love the library and how much more comprehensive it is than all the other options out there. Nice work!

Am currently working on an Arduino implementation of a CubeSat, and will be putting the little guy on a balloon launch this Saturday (a couple of days from now) and would love to be able to use your library.

It complies fine, but when uploading the following error message pops up

"Flash for chip ID 7e06e0e6 is not supported"

I'm using a 3rd party Arduino Due copy - http://www.geeetech.com/wiki/index.php/Iduino_DUE_Pro -

Any help would be much much appreciated.

Thanks,
Jordan

arkhipenko, 2015-11-17:
Hi Jordan,

As much as I would like to help (and thankful for your praise) I have never worked with Due, don't have a board like that, and unfortunately have never come across this error. 

Some troubleshooting questions:
Does it only happen when you use TaskScheduler library?
Can you load a example TaskScheduler sketch up to Due?
Based on the fact that it compiles fine, I would venture to guess library is not the culprit, but one can never be sure...

Google search shows this a rare error (I got only 3 hits)...

Sorry that I could not be more helpful.

Anatoli

jjol7166, 2015-11-17:
Hey Anatoli,

Thank you for the quick reply!

I actually just managed to overcome the bug (10mins ago) by manually resetting the board (via the hardware reset pin) in-between clicking upload and the program actually starting to upload (during the compiling stage). 
I have no idea why this works, but it works haha.

I will now be able to use you're software on the CubeSat!

For the flight, I'll be using that watchdog timer functionality you provided (as the balloon is going 50kms up and various sensors etc could freeze), as well as the IDLE_MODE to conserve battery power. 

While I do have you here though - can you think of anyway to give runtime preference to a High Priority task over a Low Priority task. For the two cases where: both tasks are requested at the same time, OR, High Priority task is request while a Low Priority task is currently being executed. 

Thanks!
Jordan

arkhipenko, 2015-11-17:
I think your issue has to do with the way Due programming sequence has to work:
Check this: (sounds relevant) https://www.arduino.cc/en/Guide/ArduinoDue#toc4

I am very excited TaskScheduler goes 50 km up! Would you mind keeping me posted on the progress? Perhaps some pictures? Please send to arkhipenko@hotmail.com

As for the priority functionality. I have been thinking a lot about it (especially since Symbian's Active Objects, which TaskScheduler is inspired by support it), but decided against it.  This is why:

1. Execution chains are simple and efficient. The main idea is to minimize scheduling overhead by Scheduler going through the chain. Implementing true priority would require looking ahead through the entire chain, then ranking and aging tasks by priorities. It would slow the execute() loop significantly.

2. TaskScheduler is NOT a pre-emptive multi-tasking library. Nor is it a Real-Time OS. There is no way to break execution of one task in favor of another. Therefore callback methods require careful programming. This has, however, significant benefits: you don't need to worry about concurrency inside the callback method, since only one callback method runs at a time, and could not be interrupted. All resources are yours for that period of time, noone can switch the value of variables (except interrupt functions of course...), etc. It is a stable and predictable environment, and it helps a lot with writing stable code. 

A number of things could be done instead of priorities:

1. Schedule your critical tasks to run more frequently than the other tasks
2. If one callback routine is critical, create a couple of tasks referring to the same callback and "sprinkle" them around the chain:

    Scheduler ts;
    Task t1(20, TASK_FOREVER, &callback1, &ts);
    Task t2(1000, TASK_FOREVER, &callback2, &ts);
    Task t3(20, TASK_FOREVER, &callback1, &ts);
    Task t4(1000, TASK_FOREVER, &callback4, &ts);
    t3.delay(10);

Note that t1 and t3 call the same callback method, and are shifted in time by 10 millis. So effectively callback1 will be called every 10 millis, but would be "sandwithced" between t2 and t4. 

3. Use short efficient callback methods written for cooperative multitasking.
What that means is:

a) DO NOT use Arduino's delay() function. It is blocking and will hold the entire chain. Instead break the callback method into two, switch the callback method of the task where delay is necessary and delay the task by that number of millis. You get your delay, and other tasks get a chance to run:

instead of:

    void callback() {
     ... stuff
     delay(1000);
     --- more stuff
    }
     
do this:

    void callback1() {
     ... stuff
     t1.setCallback(&callback2);
     t1.delay(1000);
    }
    void callback2() {
     ... more stuff
     t1.setCallback(&callback1);
    }

b) Same goes to pulseIn() function. If you have to use it, set the timeout parameter such that it is not a default 1 second. PulseIn functionality could be achieved via pin interrupts, and that solution is non-blocking.

c) Do don run long loops (for or do/while) in you callback methods. Make the main arduino loop be the loop driver for you:

instead of:

    void callback() {
    
      for(int i=0; i<1000; i++) {
        ... stuff // one loop action
      }
    }

do this:

    Task t1(TASK_IMMEDIATE, 1000, &callback);
     
    void callback() {
      int i = t1.getRunCounter() -1;
      ... stuff // one loop action
    }

or this:

    Task t1(TASK_IMMEDIATE, 1000, &callback, true, &t1On);
    
    int i;
    bool t1On() {
      i = 0;
      return true;
    }
    
    void callback() {
      ... stuff // one loop action
      i++;
    }

d) Break long running callback methods into several shorter ones, and pass cantrol from one to the other via setCallback() method:

    Task t1(TASK_IMMEDIATE, TASK_FAREVER, &callback);
    
    void callback() {
      ... do some stuff
      t1.setCallback(&callback_step2);
    }
    void callback_step2() {
      ... do more stuff
      t1.setCallback(&callback_step3);
    }
    void callback_step3() {
      ... do last part of the stuff
      t1.setCallback(&callback);
      t1.delay(1000);
    }

This will execute all three parts of the callback function in three successive steps, sheduled immediately, but allowing other tasks in the cahin to run. Notince that task is scheduled to run immediately, and 1 second period is achieved by delaying the task for 1000 millis at the last step. 
Alternatively you could schedule the task to run every 1000 millis and use forceNextIteration() method in steps 1 and 2 (but not 3!)

    Task t1(1000, TASK_FAREVER, &callback);
    
    void callback() {
      ... do some stuff
      t1.setCallback(&callback_step2);
      t1.forceNextIteration();
    }
    void callback_step2() {
      ... do more stuff
      t1.setCallback(&callback_step3);
      t1.forceNextIteration();
    }
    void callback_step3() {
      ... do last part of the stuff
      t1.setCallback(&callback);
    }

e) Compile the library with _TASK_TIMECRITICAL enabled and check if your tasks are falling behind schedule. If they are - you need to optimize your code further (or maybe re-evaluate your schedule). If they are not - all is well and you don't need to do anything. E.g., I have a spider robot which needs to measure distance, control motors, and keep track of the angle via quering gyroscope and accelerometer every 10 ms. The idea was to flash onboard LED if any of the tasks fall behind. At 10 ms interval for the gyro the LED does not flash, which means none of the tasks are blocking the others from starting on time.  Let me know if you are interested to look at the code, I will send it to you. 

Hope this helps. If you would like me to review your code and recommend cooperative changes, please send it to arkhipenko@hotmail.com or upload it to github. 

Good luck with your project!

closed 2015-11-18 by arkhipenko