arkhipenko/TaskScheduler #9: Task schedule inconsistent at start

alw1746, 2015-11-18:
Hi Anatoli,

Nice work on the task scheduler! I tested the first example and while it worked very well, the tasks did not follow the actual schedule until the fourth iteration. Attached the monitor result and example code which I modified to display interation count. Any clues? Could it be something odd with serial.println rather than the task itself?

regards,
Alex
[Scheduler_example.txt](https://github.com/arkhipenko/TaskScheduler/files/37729/Scheduler_example.txt)
[monitor.txt](https://github.com/arkhipenko/TaskScheduler/files/37730/monitor.txt)

alw1746, 2015-11-18:
Forgot to mention I ran it on an Arduino Pro Mini 5V 16Mhz.

arkhipenko, 2015-11-18:
Please post your modified code here. I think I know what causes it, but want to make sure.

https://github.com/arkhipenko/TaskScheduler/issues/9#issuecomment-157673555

arkhipenko, 2015-11-18:
Sorry, didn't realize code was attached.
This is caused by the delay (5000) statement at the end of setup(). 
Tasks were enabled before. Delay blocks there execution for 5 sec, so first 4 cycles they "catch up". 
Try without delay or enable tasks after delay.

arkhipenko, 2015-11-18:
Check the most recent example 1 please. This is how it is done there.

closed 2015-11-18 by arkhipenko