arkhipenko/TaskScheduler #11: Xtensa compiler (esp8266): Declaration of constructor does not match implementation

nerk, 2016-01-02:
Compiling TaskScheduler.h using the `xtensa-lx106-elf-g++` compiler (esp8266) fails. Apparently, it can not match the declaration of the constructor with its implementation because of mixing `boolean` and `bool` for parameter `aEnable`.

```
public:
   Task(unsigned long aInterval=0, long aIterations=0, void (*aCallback)()=NULL, Scheduler*  aScheduler=NULL, boolean aEnable=false, bool (*aOnEnable)()=NULL, void (*aOnDisable)()=NULL);
```

needs to become 

```
public:
   Task(unsigned long aInterval=0, long aIterations=0, void (*aCallback)()=NULL, Scheduler* aScheduler=NULL, bool aEnable=false, bool (*aOnEnable)()=NULL, void (*aOnDisable)()=NULL);
```

arkhipenko, 2016-01-02:
I have not yet tried to use TaskScheduler with esp8266. Very cool. Will fix the type mismatch as soon as I get back to my laptop. 
Please do tell if it works. I have the chip, just never got around to compiling fo it.
I suspect (strongly) that power management (IDLE sleep) might not work, and I am not sure how well current version will play with the WFL. 
Would be great if you could share your experience. Thanks.

arkhipenko, 2016-01-03:
Could you please test v2.0.1?

alw1746, 2016-01-03:
Fyi,I did a quick-n-dirty private port a few months back on v1.8.2 and it did work very well. I noted:

-the task constructor declaration bug which you've now fixed.
-examples 1-5 worked with some minor recode.
-example 6(power management) worked,can't remember the counts. 
-not example 7 (WDT not available in ESP Arduino).
-example 8 worked.

A diff of my code showed I made a declaration at top:

#ifndef _ESP8266
#include <avr/sleep.h>
#include <avr/power.h>
#endif

and a conditional test for ESP8266 in Scheduler::execute()

#ifdef _TASK_SLEEP_ON_IDLE_RUN
  if (idleRun && iAllowSleep) {

#ifdef _ESP8266
    const int sleepTimeMs = 1;
    ESP.deepSleep(sleepTimeMs*1000);
#else
  	set_sleep_mode(SLEEP_MODE_IDLE);
  	sleep_enable();
	  /* Now enter sleep mode. */
	  sleep_mode();
	  
	  /* The program will continue from here after the timer timeout ~1 ms */
	  sleep_disable(); /* First thing to do is disable sleep. */
#endif

arkhipenko, 2016-01-03:
Great. Thanks for sharing. I will incorporate your updates in the next version. 
I do recommend switching to the latest v2.x - I have made significant optimization of the execute loop finding a few (minor) bugs in the process.
Plus v2 now supports task prioritization.

Enjoy! Glad you find TaskScheduler suitable for your projects.

nerk, 2016-01-03:
Thanks a lot to you for this great project! This code is exactly what I was looking for.
I only tested the most basic things and they seem to work as expected, Obviously, idle sleep cannot work with the current code base and ESP8266. @alw1746 seems to have solved most issues.

I will try again once the other changes are in place. Cool stuff! :+1:

arkhipenko, 2016-01-03:
Thank you. Always good to know your work is useful for someone. :)

arkhipenko, 2016-01-06:
@alw1746 
Alex, it seems WDT might be available for ESP. Please check this:
https://github.com/sandeepmistry/esp8266-Arduino/blob/master/esp8266com/esp8266/cores/esp8266/Esp.h

arkhipenko, 2016-01-06:
@alw1746 
Alex, another note:
I don't think this code would work:

    const int sleepTimeMs = 1;
    ESP.deepSleep(sleepTimeMs*1000);

According to many articles (this for example: http://tinker.yeoman.com.au/2015/03/08/reducing-esp8266-power-consumption-using-deep-sleep/) esp8266 does a full reset after wake up, which would restart the sketch - not the behavior execute() function needs.

alw1746, 2016-01-06:
hmm..in that case I don't understand. I ran example 6 and this is what I got,
Scenario 1): With #define _TASK_SLEEP_ON_IDLE_RUN enabled
Start
c1=497878
c2=1001
Scenario 2): With #define _TASK_SLEEP_ON_IDLE_RUN disabled (commented out)
Start
c1=828914
c2=1001
C1 in scenario 2) is higher than in scenario 1) because processor is put to sleep for 1), but not for 2). But the count is only half so it's not a big diff, plus I did not see a reset. Perhaps ESP.deepsleep isn't working as I thought? Let me reflash my ESP8266 with the latest Arduino release and taskscheduler and I'll report back.

arkhipenko, 2016-01-06:
@alw1746 
Alex, Comparing your results to the 16Mhz Uno:

for the following tasks:

    Task c(10, -1, &Count, &ts);
    Task t(10000, 1, NULL, &ts, true, &tOn, &tOff);

c1 should be "around" 10000 (with the IDLE RUN SLEEP enabled), so clearly it does not sleep.
c2 should always be 1000 (10000/10) 

Without IDLE RUN SLEEP for UNO at 16 MHz c1=529783  (v1.9.0: 551947)

I suppose ESP8266 is running at 80 MHz, but is probably doing a lot more in the background for WFL, so straight extrapolation (~550000 / 16 * 80 = 2750000) will not be accurate. 828914 "feels" low, but then again, I am not sure what else CPU is scheduled to do (probably a lot). 

I guess it is time for me to pull my esp8266 out and start testing :)

arkhipenko, 2016-01-11:
@alw1746 
Hi Alex, just wondering if you had a moment to test. Not urgent, just curious to know how it works with ESP8266... Thanks.

alw1746, 2016-01-12:
sorry mate been away so haven't had a chance to make blinky lights. But I'm downloading the latest ESP8266 Arduino build now so will let you know in the next day or so...

arkhipenko, 2016-01-12:
@alw1746 no problem! Thanks for doing this actually. Please use the testing branch for this.

alw1746, 2016-01-13:
Had some problems with my ESP8266 Arduino v2.0.0 download so I tested Taskscheduler 2.02 (testing branch) against ESP8266 v1.65 on my board already. Attached test results. All worked apart from Example 7 (WDT) which I'll test again once I get 2.0.0 sorted out. Some minor mods needed, see the test results.
[TaskschedulerTestResults_ESP8266.txt](https://github.com/arkhipenko/TaskScheduler/files/88706/TaskschedulerTestResults_ESP8266.txt)

alw1746, 2016-01-13:
Running with ESP8266 Arduino 2.0.0 now but can't get Example 7 to work. Replaced the timeout setting code with,

  //reset watchdog
  wdt_reset();
  ESP.wdtEnable(WDTO_2S);
  //Enable global interrupts

But below ISR failed to compile as WDT_vect is not exposed in Esp.h. 

ISR(WDT_vect)

arkhipenko, 2016-01-13:
@alw1746
Alex, thanks for testing. Good news!

Let me look into WDT a little closer. 

And I still didn't put the chip to sleep per se, just delayed by 1 ms in case IDLE SLEEP is activated, so a bit of "cheating" on my end...
But it is entirely possible that ESP8266 does not support sleep the same way AVRs do, so maybe this is the only way. 

To be investigated further....

arkhipenko, 2016-01-13:
@alw1746 
Addressed all of your changes in the main code of the examples for the next release, except for the WDT. Will have to look into that and possible sleep next.

arkhipenko, 2016-01-13:
Just pushed the update with examples changes into 'testing' branch.

closed 2016-02-04 by arkhipenko