arkhipenko/TaskScheduler #12: Finer time resolution

ThomasCLee, 2016-02-03:
I initially used your library to do multitasking on Arduino for my robot and 3D printer. I found that the time resolution (in milli-seconds) is not fine enough for my applications. I ended up copying your library and replacing all the millis() with micros(). 

It would be nice if TaskScheduler class could be updated to support finer time resolution.

arkhipenko, 2016-02-03:
Funny you say that - please check out the "testing" branch - I am testing exactly that!

#define _TASK_MICRO_RES			// Support for microsecond resolution

before include and recompile.

ThomasCLee, 2016-02-03:
:)

Couldn't wait to see it released to the main branch.

arkhipenko, 2016-02-03:
Working on it. Would greatly appreciate if you could run some tests...
Do you have a website for your project? Very curious...

ThomasCLee, 2016-02-03:
Sure, I will test it out tonight or tomorrow.
Well... regarding my robot project... you can check out my website: http://www.hackarobot.com
However, I didn't publish anything related to multi-tasking because I want to get the official support from your library first.

Regarding my 3D printer project. I am completely re-writing Marlin from scratch using C++. I am moving a few motors at the same time and millisecond resolution is simply not good enough.  With the modified TaskScheduler class, the motors moves much more smoothly.

ThomasCLee, 2016-02-03:
I tried it. 
I included it in a simple program. It seems to work if the #include <TaskScheduler.h> is added in the main ".ino" file.

However, when I try to include it in the header file in a Object class (e.g. Motor.h), it doesn't find the library. Any idea how I can update the search path? (note, TaskScheduler.h is already placed at 

Arduino --> Libraries --> TaskScheduler --> TaskScheduler.h

arkhipenko, 2016-02-03:
What if you place #include <TaskScheduler.h> before #include <Motor.h> in the .ino?  Should have everything defined for the Motor.h to work. No?

ThomasCLee, 2016-02-03:
Well... it doesn't work even if I include TaskScheduler in .ino before anything else.

Here is how my code look...

In Motor class, I need to pass the pointer of Scheduler class so that all motors refer to the same scheduler. Therefore, I need to include TaskScheduler in Motor.h

class Motor {
 private:
  Scheduler *sch;

...

It seems to be just a problem of the search path. I don't know how to get around this.

On the other hand, if I copy your .h file, rename everything to something like "MyScheduler.h" and place the .h in the same directory where .ino file sits, it compiles just fine.

arkhipenko, 2016-02-03:
Just to confirm: 
You have two libraries: Motor and TaskScheduler, both in the 'libraries' folder of the Arduino sketch root folder; you have #include \<TaskScheduler.h\> in the Motor.h file, and it does not find it? Am I getting it correctly?
Can I look at the code (if you don't mind?)

arkhipenko, 2016-02-03:
I just ran a test:

In the TaskScheder .h file:

    #include <Arduino.h>
    #include <I2C.h>
    I2C a;

    #ifndef _TASKSCHEDULER_H_
    #define _TASKSCHEDULER_H_

and it worked fine... Found I2C.h and compiled the sketch... Which version of IDE are you on?

ThomasCLee, 2016-02-03:
No, Motor class only exists in the folder where the main .ino is located. Perhaps, I should move it to the libraries folder.

ThomasCLee, 2016-02-03:
I am using 1.6.5 IDE

arkhipenko, 2016-02-03:
I am on 1.6.7. I believe there was some significant rework on the pre-processor front.
And yes, I would recommend moving the Motor library to 'libraries' folder. That should work. 

To access current scheduler (if you compile with _TASK_PRIORITY option) you have a static method in the Scheduler class:

    static Scheduler& currentScheduler();

would that be an alternative?

arkhipenko, 2016-02-03:
By the way, I can only imagine how much fun it is to work with your robot. I built something similar, and it was a lot of fun:

http://www.instructables.com/id/Arduino-Nano-based-Hexbug-Scarab-Robotic-Spider

ThomasCLee, 2016-02-03:
Thanks. Your robot looks awesome. 
By the way, I think your library looks fine. I will move the motor class to the libraries. It would work. Closing this issue. Thanks for helping

closed 2016-02-03 by ThomasCLee

arkhipenko, 2016-02-04:
Thomas,
One question: can I reference your work in the ReadMe file for TaskScheduler?I want people to see what the library can do!
Thanks in advance.Anatoli

Date: Tue, 2 Feb 2016 20:56:58 -0800

ThomasCLee, 2016-02-04:
You are more than welcome to do so.

arkhipenko, 2016-02-04:
Thank you! Would you mind mentioning TaskScheduler somewhere there? (so people believe me.. :)

Date: Wed, 3 Feb 2016 21:19:41 -0800

ThomasCLee, 2016-02-04:
let me polish my sample code a little bit. I will let you know once I uploaded it. Of course, I will mention TaskScheduler because it is such a great library. It is essential to my application.

arkhipenko, 2016-02-04:
Well, thank you. It is very important for me to know my work is essential to other projects. 

--------

Sent from a mobile device. Apologies for accidental typos. 

From: Thomas Lee

Sent: Thursday, February 4, 11:25 AM

Subject: Re: [TaskScheduler] Finer time resolution (#12)

To: arkhipenko/TaskScheduler

Cc: Anatoli Arkhipenko

let me polish my sample code a little bit. I will let you know once I uploaded it. Of course, I will mention TaskScheduler because it is such a great library. It is essential to my application.