arkhipenko/TaskScheduler #13: Is there any way to use stateful callbacks?

adamryczkowski, 2016-03-05:
Judging from the source code, the callback is fixed as a single (function) pointer. As such, even C++11 lambda functions would not help.

Do you have any plans (and vision) on how to implement them? My analysis concluded, that the simplest and most efficient way to do it is to store the following struct

struct {
   void (* fptr)(void *);
   void* arg;
}

instead of just function pointer. Overload the `operator()` for such struct; and if the `*arg==0`, call the `fptr` without parameters.

What do you think?

arkhipenko, 2016-03-05:
Hi @adamryczkowski 

I am always open to improving the functionality of TaskScheduler, given that it maintains small memory footprint and scheduling overhead. The primary target platform is microcontrollers, which [used to] have small amount of RAM to work with, so every byte and every tick counts. 

That partially explains why everything about Tasks is so explicit - to minimize the time to get to callback functionality. Since the function pointer is explicit, and no parameters need to be pushed into stack, the callback happens relatively quickly.  Using overloaded functions will (in my opinion) add scheduling overhead, so I would like to understand what is the use case for your request?

If you would like to use the same callback code for different tasks, consider using the Local Task Storage pointer functionality. Every Task can store a pointer to its specific data structure. Callback can get a reference to current task from the scheduler and access task specific variables. There is a chapter about that in the documentation.

Hope this helps. But would be happy to continue the discussion in case I am missing some very useful way of using Tasks that is not achievable as is. 

Thanks!

PS: Are you using TaskScheduler, and if yes, what for?

adamryczkowski, 2016-03-06:
Thank you very much for such detailed answer. 

No, I didn't notice the _TASK_LTS_POINTER. I think it is everything I need for now, I'll test it today.

I am looking for a good multitasking framework that will power my future Arduino projects. The first project at hand is my "smart bathroom":

There are 2 Arduinos in my bathroom. The main one is connected to an external CD74HC4067 LED PWM controller that powers powerful 4x4 led stripes (RGB, white and red), motion detector, hall effect based opened doors detector, TCS34725 colour detector (for calibration of LED colours), real time clock, a switch that controls floor heating pump, thermometer and another bathtub Arduino that controls 3 thermometers connected to my bath tub (to measure water temperature and - indirectly - water level), two extra LED stripes + one WS2812B vertical stripe with a nice animation of fire). I also want to test infra-red pilot for more control over the modes.

In the future, beside the autonomous work, The Arduino will be connected (serially via USB, so I could re-program Arduino remotely) with Raspberry Pi, that will do voice recognition and display a nice touchscreen with StarTrek-themed control panel :-). 

So there is a loot of simultaneous work to do (lights, bathtub, heating and being managed by Pi).

The second project will be a sort of Jedi-saber for my son. It will be battery-powered, so I want to press for power-efficient design. By design the saber will have ca. 100 WS2812B neopixel stripe, accelerometer, a touch sensor and maybe a flex sensor. I want it to be capable of showing a nice picture when rotated, and show nice animation when touched/wiggled/bent.

arkhipenko, 2016-03-06:
Sounds very exciting!  Do you have a web page (or instructables page) to look at it?

adamryczkowski, 2016-03-06:
Nope. Not yet. But thinking of it - why not? I'll try, but I cannot promise when. It is just that tinkering with electronics is only my hobby, and I barely find time for it.

arkhipenko, 2016-03-07:
Yeah. Finding time is the biggest problem... I can relate. Good luck!

closed 2016-03-07 by arkhipenko