arkhipenko/TaskScheduler #14: What is the best way to use TaskScheduler with POWER_DOWN on idle (instead of the IDLE mode)?

adamryczkowski, 2016-03-06:
If you power down Arduino instead of putting it to sleep, you can wake it with a watchdog timer. I know, that you use a watchdog timer for checking for scheduler overruns, but only optionally. 

I also know, that when Arduino is powered down, it doesn't update `millis()` and PWMs are down too. But it can still respond to external interrupts and I2C interrupts, so it is sufficient for my projects.

Do you have any ideas on how to use powerdown mode instead of idle with your library? I can help you implement this.

arkhipenko, 2016-03-06:
Please check my project:  
  Interactive Halloween Pumpkin
    http://www.instructables.com/id/Interactive-Halloween-Pumpkin/

This is exactly how the pumpkin operates: powered down when idle, and awaken via interrupt from the motion and sound sensors. 

One thing I would do differently now though is: I would add a relay to cut the power off of the peripheral devices which should be powered down as well (i.e., the mp3 board in the pumpkin case), and power then back up when arduino wakes up. 

But conceptually, you should be able to reuse parts of the sketch which powers the device down.

arkhipenko, 2016-03-06:
However, if the question was if you can use POWER_DOWN between scheduling runs, I played with the idea long time ago and dropped it:
1. Without real time clock, there is no way to determine how long you were "asleep" (millis doesn't help as you pointed out). You could be down for 8 seconds, or wake up in 3 seconds on external interrupt. 
2. Restoring from powerdown is more involved, as most of the components were down too (serial, etc), and it takes another ~50 ms to fully power up, so doing it every 8 seconds was a bit too much,

adamryczkowski, 2016-03-06:
> 2. Restoring from powerdown is more involved, as most of the components were down too (serial, etc), and it takes another ~50 ms to fully power up, so doing it every 8 seconds was a bit too much

Are you sure about these 50ms? It seems you are quoting the reset time, not waking from POWER_DOWN. [http://forum.arduino.cc/index.php?topic=142821.0](url)

arkhipenko, 2016-03-06:
There is this post which talks about different power modes:

http://www.gammon.com.au/forum/?id=11497

please search for "Warning about wake-up times" for details.

closed 2016-03-07 by arkhipenko

adamryczkowski, 2016-03-08:
Thank you very much for the link. It seems I'd need to measure it myself to settle the matter. And I think it will not be that difficult. I'll give you my test results, when they are ready.