arkhipenko/TaskScheduler #18: <avr/wdt.h>

Fbo06, 2016-07-22:
Hello, first at all congratulations and thank you for your hard work ! 
i have a problem to compile the 7th example who use #include <avr/wdt.h>
it say : "No such file or directory"
can you tell me, please, how can i do to make it work ?

arkhipenko, 2016-07-22:
Hi.

I will currently on vacation, but will look into this on Sunday.

Could you please make sure you install the library into appropriate folder with appropriate folder structure?

Also, please let me know which version of Arduino GUI do you use, and which target device do you compile for? (Not all devices support WDT).

Sent from a mobile device. Apologies for accidental typos.

https://github.com/arkhipenko/TaskScheduler/issues/18

Fbo06, 2016-07-22:
Hi Anatoli,
Thank you to respond me so quickly 
- i install the library by the Arduino GUI
- Arduino GUI is 1.6.8
- The Target Device is : WeMos D1 R2 & Mini
All the Samples work perfectly, only this one using "#include <avr/wdt.h> » don’t want to compile
Take care.
Best regards.
Frederic.

arkhipenko, 2016-07-30:
Hi,
WeMos D1 R2 is ESP8266 based controller. ESP8266 is not an avr board, therefore avr/wdt.h is not defined for it.
ESP8266 has a watchdog timer, but it is different.
Sorry I cannot help you with this - have not worked with ESPs enough to know.

closed 2016-07-30 by arkhipenko

Fbo06, 2016-07-31:
Hi Anatoli Thank you for your research, i will look for this watchdog.

arkhipenko, 2016-07-31:
Here is a link that seems to address your question:

http://www.esp8266.com/viewtopic.php?f=32&t=2776

arkhipenko, 2016-07-31:
Check this post. Seems relevant:
http://www.esp8266.com/viewtopic.php?f=32&t=2776

Date: Sat, 30 Jul 2016 23:49:31 -0700

Fbo06, 2016-07-31:
Thank you, i also have a link https://github.com/esp8266/Arduino/issues/1532 and i try it it's work but i don't know how to save the status of pin to restore them after re booting.

arkhipenko, 2016-07-31:
Could you describe what you are trying to achieve?

Sent from a mobile device. Apologies for accidental typos.

https://github.com/arkhipenko/TaskScheduler/issues/18#issuecomment-236443651