arkhipenko/TaskScheduler #19: Arduino Mega AT2560

lv2eof, 2016-08-10:
Hi,
Thanks for your work.
I'm trying to run some examples in a Arduino Mega AT2560 with a Linux computer.
They compile well, upload well but in the Serial Monitor window I get a lot of strange characters.
Can you give me please an idea about what this could be?
Thanks in advance,
Luis

arkhipenko, 2016-08-10:
Hi Luis,

Sure. Please check that the serial baud rate of the demo sketch (I believe I consistently used 115200) matches the baud rate of the terminal window. 

Look for Serial.begin (115200);

Also, I never had a chance to test on Megan, so expect unexpected ;)  However, I ran TaskScheduler based sketches on Micro, which I believe is a close cousin, so you should be able to use it for mainstream cases. 

Good luck!  If not a secret, what are you using TaskScheduler for?

Sent from a mobile device. Apologies for accidental typos. 

From: Luis Vital

Sent: Wednesday, August 10, 4:20 PM

Subject: [arkhipenko/TaskScheduler] Arduino Mega AT2560 (#19)

To: arkhipenko/TaskScheduler

Hi,

Thanks for your work.

I'm trying to run some examples in a Arduino Mega AT2560 with a Linux computer.

They compile well, upload well but in the Serial Monitor window I get a lot of strange characters.

Can you give me please an idea about what this could be?

Thanks in advance,

Luis

lv2eof, 2016-08-11:
Hi Anatoli,
Thanks.
Now I can read what comes out :)
Next step will be try to understand a bit how things work.

Regarding why I'm using it:
I'm just beginning.
In school I have to do something with FreeRTOS so I google a bit around.
Your project is a bit out of the scope of my school work, but maybe I can
get some ideas.

Anyway it seems a nice work you have been doing, congratulations :)
Cheers,
Luis

closed 2016-08-11 by arkhipenko