arkhipenko/TaskScheduler #25: Also for ESP8266?

ghislainborremans, 2017-02-21:
Hello, 
is this only for Avr arduino or can it be used with ESP8266

best regards

arkhipenko, 2017-02-21:
It's a good point though: I should make it clear that this library supports esp8266.

ghislainborremans, 2017-02-22:
Just add that the example07 is not working on ESP8266 because of "avr/wdt.h"

closed 2017-02-23 by arkhipenko

abhishek2184, 2019-02-21:
Hi,
I uploaded the yield code, given in the example(Scheduler_example14_yield.ino), to my AI Thinker ESP01 module. I am not able to connect to my router, even after several retries. I know that the code is tested for node mcu, but I don't think it should not work for Generic ESP8266 modules.

Thanks

arkhipenko, 2019-02-21:
Could you please open a new issue? Closed issues are difficult to track.
Also, I assume, but would like to confirm that you have changed these lines:
``` c++
// Replace with WiFi parameters of your Access Point/Router:
const char *ssid  =  "wifi_network";
const char *pwd   =  "wifi_password";
```
to the appropriate SSID and a password, and your router are not running 5GHz WiFi because I believe esp8266 does not connect to those.

abhishek2184, 2019-02-22:
Thanks for the reply, I have opened a new issue "Cannot connect to Access Point". I have also uploaded the modified example in that thread. Please see if you can find the problem.
With regards