arkhipenko/TaskScheduler #26: interrupt

ghislainborremans, 2017-03-18:
Hello, 
Is it possible ( advisable) to have a interrupt attached in "loop" for a sensor measurement or will this cause problems with the task scheduler?

arkhipenko, 2017-03-19:
Hi,

I need to understand what you are trying to do a little better to opine.

If you are asking whether you could put some code into loop() method together with scheduler's execute() method, then the answer is yes you can, but I would use a task for it.

If the question is about pin interrupts, then there are two ways to use tasks:
Create a task that runs once immediately, but make is disabled at creation. In the pin interrupt's ISR call tasks restart() method. The task will run at the next pass.

Another approach is to create a StatusRequest object and a task that waits on its completion.
In the interrupt's ISR complete the StatusRequest object, and the task which is waiting on it will run during the next pass.

Both ways are valid, the second in more robust, as you can trigger multiple (and changing) number of tasks as a result of an interrupt.

Also, this way of dealing with interrupts is preferable because you exit ISR quite quickly, which is how it should be done.

If you want, send me your code and I will have a look to see how you could take more advantage of taskscheduler.

Cheers, and thanks for kind words about my library. Much appreciated.

Regards
Anatoli

Sent from a mobile device. Apologies for accidental typos.

ghislainborremans, 2017-03-19:
Hello Anatoli
perfect explanation. I will give it a try this way. 

Do i understand this correctly: i have task 1,2,3,4,5
task 3 is awaiting statusrequest. If in task4 processing time, the ISR completes the statusobject, then task5,1 and 2 will be executed and then task 3. Is there a way to have task3 executed after task4 ( except to mingle task 3 all around several times?

arkhipenko, 2017-03-19:
Multitasking is interesting this way. In your example, the tasks will be evaluated in order 5,1,2,3 for execution, but it does not mean they will run in that order. They only run if they are ready.  The task that was waiting on a StatusRequest will run during next pass for sure. 

There is a way to make task3 run after 4 if you use TaskSchedule layered prioritization. The documentation describes how to do it: you need to create 2 schedulers, link them, and have task 3 attached to  higher priority scheduler, while all other tasks are attached to lower priority scheduler. Task 3 will be evaluated for running after every task of the lower priority scheduler. 

Check the documentation please, there is also an example I believe, and my hexbug code uses layered prioritization (link is in the readme).

arkhipenko, 2017-03-19:
Here is the code from scarab. Gyro scheduler is a higher priority one:

//  taskManager.init();

  gyroManager.addTask(tGyro);
  gyroManager.addTask(tAccel);
  
  taskManager.addTask(tDistance);
  taskManager.addTask(tMove);
  taskManager.addTask(tMoveCheck);
  taskManager.addTask(tBlinkRed);
  taskManager.addTask(tBlinkYellow);
  taskManager.addTask(tBlinkGreen);
  taskManager.addTask(tError);
  
  taskManager.setHighPriorityScheduler(&gyroManager);

closed 2017-03-23 by arkhipenko