arkhipenko/TaskScheduler #27: Run counter as an index into array within callback.

alw1746, 2017-03-27:
G'day mate, I have an issue using run counter as an index into an 8-byte array inside a callback. If the callback invokes another callback, run counter=1,3,5,7... instead of 1,2,3,4,...(see attached sketch). My workaround is to increment a global index variable and set iterations*2 (8*2). But is there a a more elegant way to achieve this?

[testtask.ino.txt](https://github.com/arkhipenko/TaskScheduler/files/872121/testtask.ino.txt)

arkhipenko, 2017-03-27:
Well, based on the sketch, everything works as designed.
The scheduler does not know that you are switching callback methods at every iteration. The switch of the callback method does not mean that it will be invoked immediately - it will be invoked at the next pass, when the counter is increased by 1.

Using global variable is the simplest and easiest solution. 

Knowing that there are only 2 possible callbacks, and they always alternate, you could do this:
    int counter=( t1.getRunCounter() - 1) >> 1 + 1; 
That should translate 1,3,5,7 into 1,2,3 4, and 2,4,6,8 into 1,2,3,4 as well

The more extravagant solution would be to use the local task storage pointer. I would not do it for the simple cases, but if you have variable number of tasks, that could be a way to go. 
Scheduler_example08_LTS explains how to use LTS. It comes with a bit of an overhead at every iteration, but is guaranteed to be related to a specific task object, so you can increment appropriately.

Hope this helps.

alw1746, 2017-03-28:
Cool, that's good enough for me. cheers....alex

closed 2017-03-28 by alw1746

arkhipenko, 2017-03-28:
What are you using TaskScheduler for (if not a secret)?

alw1746, 2017-03-29:
No secret mate, I'm developing a simple game (blow up bricks in wall) on an arduino using a 16x2 LCD. Your taskscheduler is very useful as I need to check button and pot,drop/shoot bricks,animate bullets,play sounds,etc. Software and schematic will be open-sourced once I finish debugging it. Everyone needs to write a game, lots of fun!