arkhipenko/TaskScheduler #28: std::function support

BlackEdder, 2017-04-07:
For one of my projects I preferred to use `std::function`s instead of function pointers. I have made the needed changes to the source code [here](https://gitlab.com/BlackEdder/painlessMesh/blob/scan_unknown_nodes/src/painlessScheduler.h). If you are interested in adding support `std::function` to TaskScheduler, I'd be happy to clean it up and submit a PR. The needed changes are fairly straightforward.

arkhipenko, 2017-04-07:
Interesting. And thank you for working on this.
I have not had a need to use std::functions. Could you please elaborate why this is needed?
Have you tried to compile the examples with your modified code? Did they work?
I am concerned with backwards compatibility obviously...

BlackEdder, 2017-04-07:
As far as I know there is no _need_ to use them over the current use, I just prefer them, because in my opinion it helps with readability:

- Can use anonymous functions.
- You can define the function right where you are using it (readability).
- You can easily pass variables along (readability).

`std::function` is more general, although at the cost of a slight overhead. Note though that for example os_timer of esp8266 takes std::function objects, so they seem to think the overhead is insignificant enough on their hardware.

```C++
typedef struct {
  unsigned int address;
  unsigned long distance;
} sensor_data;

sensor_data s1;

void Measure() {
  Sensor_data& V1 = *((sensor_data*) ts.currentLts());

  V.distance = <calculate your values here>;
}

void setup() {
  s1.address = some_address;
  s1.distance = some_distance;
  ...
  t1.setLtsPointer(&s1);
  t1.setCallback(&Measure);
}
```

With std::function it can be much more compact while carrying the same information:

```C++
void setup() {
  t1.setCallback([&some_address, &some_distance]() {
      some_distance = <calculate your values here>;
  });
}
```

About backward compatibility: std::function is designed to be backward compatible with receiving a function pointer. I have not explicitly tested this for TaskScheduler, but if you are seriously considering supporting this then I would test all examples before submitting a PR.

arkhipenko, 2017-04-18:
Sorry - was on vacation (Sprint break). Will look into this later this week/ next week.

arkhipenko, 2017-04-28:
Hi Edwin,

Not sure you saw - I have merged your updates into the "testing" branch.

I will be recompiling and testing my existing code with your changes to make sure everything is backwards compatible.

Would you be willing to write a test code/example for the use of std::functions?  Ideally something that showcases the true power of that functionality.

Your help is greatly appreciated.

Thanks,

Anatoli

BlackEdder, 2017-04-28:
Of course. Does any of the current examples use the data pointer? Might be easiest if I start from that.

Should I submit it as a separate PR? Part of this one? Or just copy and paste into this issue?

arkhipenko, 2017-04-28:
Hi Edwin

Appreciate the help.

The local pointer functionality is

 *  #define _TASK_LTS_POINTER       // Compile with support for local task storage pointer

Scheduler_example08_LTS

Separate PR or direct email to arkhipenko@hotmail.com  - whatever works better for you.

By the way, do you mind if I rename the callback types?

callback_t

onDisable_cb_t

onEnable_cb_t

are out of style of the rest of the library...

Thinking:

TaskCallback

TaskOnEnable

TaskOnDisable

Thanks,

Anatoli

BlackEdder, 2017-04-28:
> By the way, do you mind if I rename the callback types?

Of course not. I was not that happy with the names anyway :+1:

arkhipenko, 2017-06-06:
I ran into a problem with testing of callback methods refactoring. Might need your help there. Details later.

arkhipenko, 2017-08-31:
Fixed in testing branch. Would you please test/confirm?

arkhipenko, 2017-08-31:
Hi, I am testing the new callback method types, and getting this warning (on Uno) and hard error (on ATTiny85) during compilation. What am I doing wrong?

> 
Build options changed, rebuilding all
Scheduler_example17_desrtructor:19: error: invalid conversion from 'void (* (*)())()' to 'TaskCallback {aka void (*)()}' [-fpermissive]
 Task tMain(TASK_SECOND, 10, MainLoop, &ts, true); 
                                                ^
In file included from /mnt/nfs/omega/home/anatoli/sketchbook/TaskScheduler/Scheduler_example17_desrtructor/Scheduler_example17_desrtructor.ino:11:0:
/mnt/nfs/omega/home/anatoli/sketchbook/libraries/TaskScheduler/src/TaskScheduler.h:195:1: error:   initializing argument 3 of 'Task::Task(long unsigned int, long int, TaskCallback, Scheduler*, bool, TaskOnEnable, TaskOnDisable)' [-fpermissive]
 Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {

 ^
/mnt/nfs/omega/home/anatoli/sketchbook/TaskScheduler/Scheduler_example17_desrtructor/Scheduler_example17_desrtructor.ino: In function 'void (* MainLoop())()':
Scheduler_example17_desrtructor:30: error: invalid conversion from 'void (* (*)())()' to 'TaskCallback {aka void (*)()}' [-fpermissive]
   Task *t = new Task(p, j, Iteration);
                                     ^
In file included from /mnt/nfs/omega/home/anatoli/sketchbook/TaskScheduler/Scheduler_example17_desrtructor/Scheduler_example17_desrtructor.ino:11:0:
/mnt/nfs/omega/home/anatoli/sketchbook/libraries/TaskScheduler/src/TaskScheduler.h:195:1: error:   initializing argument 3 of 'Task::Task(long unsigned int, long int, TaskCallback, Scheduler*, bool, TaskOnEnable, TaskOnDisable)' [-fpermissive]
 Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {

 ^
exit status 1
invalid conversion from 'void (* (*)())()' to 'TaskCallback {aka void (*)()}' [-fpermissive]

Here is the sketch:
`/** 
 *  TaskScheduler Test sketch - test of Task destructor
 *  Test case: 
 *    Main task runs every 1 second and optionally generate a task object
 *    which runs 1 to 10 times, then is destroyed
 */

#define _TASK_WDT_IDS // To enable task unique IDs
#define _TASK_SLEEP_ON_IDLE_RUN  // Compile with support for entering IDLE SLEEP state for 1 ms if not tasks are scheduled to run
#define _TASK_LTS_POINTER       // Compile with support for Local Task Storage pointer
#include <TaskScheduler.h>

Scheduler ts;

// Callback methods prototypes
TaskCallback MainLoop();

// Statis task
Task tMain(TASK_SECOND, 10, MainLoop, &ts, true); 

TaskCallback MainLoop() {
  Serial.print("MainLoop run: ");
  int i = tMain.getRunCounter();
  Serial.println(i);

  // Generating a task
  long p = random(100, 1001); // from 100 ms to 1 second
  long j = random(1, 11); // from 1 to 10 iterations)
  Task *t = new Task(p, j, Iteration);
  ts.addTask(*t);
  t->enable();
}

TaskCallback Iteration() {
  
}

/**
 * Standard Arduino setup and loop methods
 */
void setup() {
  Serial.begin(115200);

  randomSeed(analogRead(0)+analogRead(5));
 
  Serial.println("Dynamic Task Destructor test");

}

void loop() {
  ts.execute();
}`

BlackEdder, 2017-11-02:
> TaskCallback MainLoop();

I don't think this works for C function definitions, because unnamed functions are not supported. Note that for `std::function` you also need a special signature `TaskCallback mainLoopFunction = []() { ... };`. 

The closest you might get is to create a mainLoopFunction as follows (untested):

```
void MainLoop() {
  ...
}

TaskCallback mainLoopFunction = &MainLoop;
```

or the following might also work, but did not test it:

```
TaskCallback MainLoop;

void MainLoop() {
  ...
}
```

arkhipenko, 2017-12-12:
Implemented in v2.5.0

closed 2017-12-12 by arkhipenko