arkhipenko/TaskScheduler #35: Blocking tasks request

syproduction, 2017-06-02:
Hi!
Can I do a program executing one task after another? For example, create a task to execute for a particular time and other tasks are waiting?

arkhipenko, 2017-06-02:
Of course!
Just enable the next task in the OnDisable method of the previous.

Sent from a mobile device. Apologies for accidental typos.

arkhipenko, 2017-06-02:
The other way is to use StatusRequest objects and trigger subsequent task execution when the appropriate task finishes. StatusRequest objects are described in the docu and there are a couple of examples with them. Also, my latest IoT_APIS project uses them a lot to coordinate readings from multiple sensors -  feel free to have a look!

closed 2017-06-02 by arkhipenko