arkhipenko/TaskScheduler #36: Using the TaskScheduler inside another library

wvmarle, 2017-06-02:
Hi,

I'm trying to turn my working script into a library, making it much easier to reuse for other - very similar - projects. The TaskScheduler works perfectly from my original sketch, but I can't get it to compile when moving into a separate library object. See also [my post on the Arduino forum](http://forum.arduino.cc/index.php?topic=481103.msg3285271) for code example.

Compiler says:
`error: '((HydroMonitorMaster*)this)->HydroMonitorMaster::tReadSensors' does not have class type tReadSensors.enable();`

Probably more of a generic programming issue than a TaskScheduler issue, I haven't been able to find a solution for this.

arkhipenko, 2017-06-02:
I dont think I ever intended TaskScheduler to be used _inside_ another library. I am not even sure how this would work given that TS is the orchestrator of all tasks, and the execute method of the scheduler should ideally be the only thing running in the loop() method of the sketch.   

TS is not just a library, it is a framework of developing sketches - where process is broken into tasks and schedule, and is orchestrated by the scheduler. I am not sure you can easily mix it with traditional way of setup()/loop() programmin, or hide it inside a library. Definitely cannot guarantee the scheduling part if other instructions are running in the loop() and delaying the execution of main scheduling passes..  

If you send me your source code directly to arkhipenko@hotmail.com, I can take a look.

wvmarle, 2017-06-02:
Thanks for the comment. Right on time as after refactoring the rest I am back at the TaskScheduler part.

Basically what I did is take my existing sketch and turn the whole thing into a library. Most parts were split off already, now it was the last part which includes task scheduler. Maybe it's not possible what I try, or simply not a good idea. I did intend to keep my loop() pretty much empty, like the normal `ts.execute()` call there.

If keeping the TaskScheduler in my sketch (it's in the end just a few lines of code) would it work to have the tasks be functions of an external class? E.g.:

`Task tReadSensors(10*1000, TASK_FOREVER,  &readSensors, &ts, false);`

where `readSensors` is not in my .ino but an external class, that's `#include`d in the sketch. Those `Task`s can be set up in the `setup()` of my sketch, together with a `TaskScheduler` which would then be called in the `loop()`.

And related, would it be possible to pass a Task reference into the library (e.g. through the constructor, or something like `sensors.begin(tReadSensors)` ), so that over there I could `enable()` or `disable()` the task as needed?

wvmarle, 2017-06-02:
For the source: I do intend to open source it and upload it to GitHub (and get some nice version control in the process), just never got around to actually do it as it's still changing so drastically.

arkhipenko, 2017-06-02:
Still not sure what you are doing, so need to see the code before I can tell whether it's possible or not.

Definitely have not been intended that way though...

arkhipenko, 2017-06-06:
Please let me know when the code is available online. I will close this issue for now.

closed 2017-06-06 by arkhipenko

wvmarle, 2017-06-06:
I will post it - hopefully soon - working out the last kinks, then some more testing to do. I'll let you know when ready.
The TaskScheduler is back to the main sketch and out of the libraries, too much of a headache that was causing. Given up on my original design ideas.

arkhipenko, 2017-06-06:
Remember, it was written for Arduino Uno and Nano with 16K of code memory. I was not thinking about making it a proper OO library - rather a small and efficient scheduler. Hope this philosophy shows through the design.
Obviosly the ESP8266 with 100's of KB of memeory space is a different story... but stil

mianos, 2018-03-29:
I know this is closed but maybe what I just did might help the OP.
I wanted a task in a library. Here is an example of a fully isolated external class that creates its own tasks.
https://gist.github.com/mianos/51458a8113f4b03a40cb7aab7ec5430b

arkhipenko, 2018-03-29:
Thank you for your reference.
Since this issue I have created this:

https://github.com/arkhipenko/TaskScheduler/tree/testing/examples/Scheduler_example20_StatusRequest_LTS_WDT_Timeout_Object

Which I believe does what you suggested,
 and then implemented version 3 (which is in testing now) which allows dynamic binding of callbacks, putting the issues at rest forever 😊(hopefully)

https://github.com/arkhipenko/TaskScheduler/tree/testing/examples/Scheduler_example21_OO_Callbacks

Feedback welcome!

Thanks again
Anatoli

https://gist.github.com/mianos/51458a8113f4b03a40cb7aab7ec5430b