arkhipenko/TaskScheduler #37: Adding the same task twice leads to errors.

BlackEdder, 2017-06-08:
If you add the same task twice that will lead to the task chain being broken. It is possible to check against this by adding the following [here](https://github.com/arkhipenko/TaskScheduler/blob/master/src/TaskScheduler.h#L629)

```CPP
     if (aTask.iScheduler == this)
         return;
```

BlackEdder, 2017-06-08:
Similarly it might be good in deleteTask to check if the task to be deleted is actually part of the scheduler. Also during deletion set `aTask.iScheduler = NULL`. Although this would mean you might need to delete high priority task differently (check that they are part of the high priority scheduler).

arkhipenko, 2017-08-31:
Fixed in testing branch. Would you please test/confirm?

arkhipenko, 2017-12-12:
Implemented in v2.5.0

closed 2017-12-12 by arkhipenko