arkhipenko/TaskScheduler #39: Insure all examples can be compiled with continuous integration

scls19fr, 2017-09-21:
Hello,

it could be a good idea to insure that examples can be compiled successfully.

http://platformio.org/ can help as it provide continuous integration (with Travis)

http://docs.platformio.org/en/latest/ci/travis.html

Unit testing is also possible 

Kind regards

arkhipenko, 2017-09-21:
As far as I know all examples do compile successfully on their respective platforms. Which one do you have a problem with?

Sent from a mobile device. Apologies for accidental typos.

http://docs.platformio.org/en/latest/ci/travis.html

scls19fr, 2017-09-21:
I haven't test them... I'm planning to use your library. Unit testing / CI is just a good practice.
Being able to compile Arduino code on Travis is very interesting because when PR are send you can see if it doesn't break anything.

arkhipenko, 2025-09-21:
All examples are compiled and 99% of the functionality is now tested via CI/CD

closed 2025-09-21 by arkhipenko