arkhipenko/TaskScheduler #44: Stable State for an task

Jomelo, 2018-01-06:
Hello, 
it is possible to set a task in a stable state when the state is set to disable ?

For example: a pump control task

first iteration:
  // init something

loop:
 // ....

stable stop/disable state:
 //  pump stop

arkhipenko, 2018-01-06:
Yes, this is what OnEnable and OnDisable methods are for.
Please refer to [wiki-page](https://github.com/arkhipenko/TaskScheduler/wiki/API-Task#taskunsigned-long-ainterval-long-aiterations-void-acallback-scheduler-ascheduler-bool-aenable-bool-aonenable-void-aondisable)

arkhipenko, 2018-01-06:
Also, an example of use: [APIS code for watering](https://github.com/arkhipenko/apis/blob/master/apis.ino#L637)

closed 2018-01-06 by arkhipenko