arkhipenko/TaskScheduler #45: Code that worked with 2.4.0 doesn't work with 2.5.0

Wferr, 2018-01-08:
Hello, thanks for making this, it has been a lot of help for a couple projects im working on with the esp8266. Recently with the update to 2.5 code that has been working stopped compiling, I think this is due to the new multi tab compatibility but i'm not sure, any help would be gladly accepted. 

The error is below.

`
In file included from Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskScheduler.h:125:0,

                 from defcon-badge\defcon-badge.ino:20:

Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:94:3: error: conflicting declaration 'typedef struct __task_status __task_status'

 } __task_status;

   ^

In file included from Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessMesh.h:6:0,

                 from defcon-badge\defcon-badge.ino:14:

Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessScheduler.h:221:3: error: '__task_status' has a previous declaration as 'typedef struct __task_status __task_status'

 } __task_status;

   ^

In file included from Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskScheduler.h:125:0,

                 from defcon-badge\defcon-badge.ino:20:

Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:99:7: error: redefinition of 'class Task'

 class Task {

       ^

In file included from Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessMesh.h:6:0,

                 from defcon-badge\defcon-badge.ino:14:

Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessScheduler.h:230:7: error: previous definition of 'class Task'

 class Task {

       ^

In file included from Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskScheduler.h:125:0,

                 from defcon-badge\defcon-badge.ino:20:

Arduino-B\portable\sketchbook\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:187:7: error: redefinition of 'class Scheduler'

 class Scheduler {

       ^

In file included from Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessMesh.h:6:0,

                 from defcon-badge\defcon-badge.ino:14:

Arduino-B\portable\sketchbook\libraries\painlessMesh-master-1bab045503bff21f5e7b43ea120912e227835346\src/painlessScheduler.h:319:7: error: previous definition of 'class Scheduler'

 class Scheduler {

       ^
`

arkhipenko, 2018-01-08:
Is there a chance you could send me the code via PM (to arkhipenko@hotmail.com)? 
Would love to compile locally and see what causes the issue.

arkhipenko, 2018-01-08:
Also, please check the multi-tab example - there are some tricks you need to do for the multi-tab to work properly. Does that help?

Wferr, 2018-01-08:
Im using multiple tabs but with them all named as .ino so it doesn't really apply. Im working on simplifying the code im using to be just the task to help narrow down the issue currently.

arkhipenko, 2018-01-10:
Sent details via email. 
The issue is caused by the painlessMesh implementation of painlessScheduler, which is conflicting with current (and most probably all future) versions of the TaskScheduler. 
The workaround is to use painlessScheduler Tasks. (which is essentially TS v 2.3.0)

closed 2018-01-10 by arkhipenko

Wferr, 2018-01-10:
Thanks so much for the help figuring this out.