arkhipenko/TaskScheduler #51: Multiple definition error when including taskscheduler (from a separate header file.)

BlackEdder, 2018-03-03:
I tried setting up a simple project (using platformio) and get the following compilation error (I am pretty sure this used to work without a problem.)

```
ndexing .pioenvs/esp32dev/libd60/libTaskScheduler_ID721.a
Archiving .pioenvs/esp32dev/libFrameworkArduino.a
Indexing .pioenvs/esp32dev/libFrameworkArduino.a
Linking .pioenvs/esp32dev/firmware.elf
.pioenvs/esp32dev/src/main.ino.o: In function `Task::isEnabled()':
/home/edwin/TmpSync/tmpPain/.piolibdeps/TaskScheduler_ID721/src/TaskScheduler.h:308: multiple definition of
 `Task::isEnabled()'
.pioenvs/esp32dev/src/foo1.o:/home/edwin/TmpSync/tmpPain/.piolibdeps/TaskScheduler_ID721/src/TaskScheduler.
h:308: first defined here
.pioenvs/esp32dev/src/main.ino.o: In function `Task::getInterval()':
main.ino.cpp:(.text._ZN4Task11getIntervalEv+0x0): multiple definition of `Task::getInterval()'
.pioenvs/esp32dev/src/foo1.o:foo1.cpp:(.text._ZN4Task11getIntervalEv+0x0): first defined here
.pioenvs/esp32dev/src/main.ino.o: In function `Task::getIterations()':
main.ino.cpp:(.text._ZN4Task13getIterationsEv+0x0): multiple definition of `Task::getIterations()'
.pioenvs/esp32dev/src/foo1.o:foo1.cpp:(.text._ZN4Task13getIterationsEv+0x0): first defined here
.pioenvs/esp32dev/src/main.ino.o: In function `Task::getRunCounter()':
main.ino.cpp:(.text._ZN4Task13getRunCounterEv+0x0): multiple definition of `Task::getRunCounter()'
.pioenvs/esp32dev/src/foo1.o:foo1.cpp:(.text._ZN4Task13getRunCounterEv+0x0): first defined here
.pioenvs/esp32dev/src/main.ino.o: In function `Task::setCallback(void (*)())':
main.ino.cpp:(.text._ZN4Task11setCallbackEPFvvE+0x0): multiple definition of `Task::setCallback(void (*)())
'
...
```

While debuggin this I found that it happens even with just four files:

src/foo.h
```c
#ifndef FOO1_H
#define FOO1_H

#include "TaskScheduler.h"

int sum(int a, int b);

#endif
```

src/foo.cpp
```c
#include "foo1.h"

int sum(int a, int b) {
    return a + b;
}
```

src/main.ino
```c
#include "foo1.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
    sum(100, 100);
}
```

platformio.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = TaskScheduler
```

arkhipenko, 2018-03-04:
You need to do it as described in this example:
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example16_Multitab

BlackEdder, 2018-03-08:
Thanks @arkhipenko We have now solved this by including "TaskSchedulerDeclarations.h" header in most files, not the "TaskScheduler.h".

closed 2018-03-08 by BlackEdder

arkhipenko, 2018-03-16:
Check the new release I just pushed into testing branch:

https://github.com/arkhipenko/TaskScheduler/tree/testing/examples/Scheduler_example21_OO_Callbacks

This version 3.0.0 support dynamic callback method binding via compilation parameter _TASK_OO_CALLBACKS

In essence, you create each type of task as a separate class, inherited from the Task class, overload Callback(), OnEnable() and OnDisable() methods (latter 2 optionally), and have full access to your task context without having to deal with LTS.

Cheers,
Anatoli

Closed #51<https://github.com/arkhipenko/TaskScheduler/issues/51>.