arkhipenko/TaskScheduler #53: Support of Stm32duino (STM32F103C)

NickNifontov, 2018-04-03:
Can you add support of Stm32duino?

http://www.stm32duino.com/
http://wiki.stm32duino.com/index.php?title=Main_Page

arkhipenko, 2018-04-03:
It should work on it for the most part. What's probably not going to do anything is Sleep On Idle because every chip has a different way of entering power saving idle mode.
Please try to compile examples and post error messages if any.

GitMoDu, 2018-05-22:
I am unable the run the OO example. Static callbacks seem to work ok.

GitMoDu, 2018-05-24:
I've yet to use static functions with schedulers since I stopped playing in all globals and moved a library approach: hence the need for a OO centered scheduler. No more static instances just to access some variable in memory, all is contained within a few inherited member functions.

Now I just pass the scheduler pointer during instancing, and a library can have an arbitrary number of "processes" to run inside of it, so there's zero coupling between the project and the library.

So my suggestion would be to keep the separated approach: the declarations are already very complicated as it is.

My project is nearing "good enough for github" soon, keep a watch and you'll see it. I make extensive use of the scheduler to do all kinds of things, like firing a static function asynchronously from an interrupt, keep a low priority health checker running in the background and, most importantly for me, have a way to do stuff in a class and then wait an arbitrary amount of time before the next run. As long as my classes behave (more than 500us processing is the limit I accept), then it's cooperative scheduler magic in action. I already tried implementing the same stuff (successfully) over FreeRTOS (with an [appropriate OO wrapping)](https://github.com/GitMoDu/FreeRTOScpp), but the overhead was tremendous and the Heap massaging is just noise when developing.

arkhipenko, 2018-05-25:
Can I close this issue?
I do not have Stm32duino and am not planning on getting one.
If you could create a PR with the code to put Stm32duino to sleep, that'd be great.
Here is the point to do it:

https://github.com/arkhipenko/TaskScheduler/blob/6501258c99db5d1f481c098f20dc6535e3839c19/src/TaskScheduler.h#L891

GitMoDu, 2018-05-25:
I can do a few tests, but the OO example is currently not even compiling, so I wouldn't call it closed. I can make PR on that.

arkhipenko, 2018-05-25:
You mean this one does not compile?
https://github.com/arkhipenko/TaskScheduler/tree/testing/examples/Scheduler_example21_OO_Callbacks

On which board?
(And please do confirm you are using the testing branch)

It compiles for me on Uno/Nano and Teensy.

GitMoDu, 2018-07-18:
> If you could create a PR with the code to put Stm32duino to sleep, that'd be great.

I'm implementing the sleep functionality for my projects, after some testing I'll provide a pull-request, no problem.

> You mean this one does not compile?

Yes. [I found out that there is a way to check free memory in Stm32duino](http://www.stm32duino.com/viewtopic.php?f=18&t=2065), I'll see if I can easily add that so that your example remains intact.

GitMoDu, 2018-07-26:
I've implemented the suggested changes. I'll need to do some more testing, before creating a PR or two.

https://github.com/GitMoDu/TaskScheduler

GitMoDu, 2018-07-26:
The sleep on idle seems to be working fine. I have yet to test power consumption to validate, but since this is the test branch, I'm confident in merging.

As for the OOP example, I still get a segfault when iCalc is enabled in  iCalc->waitFor(iMeasure), even when I've stripped down the iCalc to do nothing. Any ideas? My simpler (in terms of scheduler feature use) projects have been working just fine otherwise.

arkhipenko, 2018-07-26:
> The sleep on idle seems to be working fine. I have yet to test power consumption to validate, but since this is the test branch, I'm confident in merging.
You can use my example 6 to validate without power consumption measurement. The counts with and without `#define _TASK_SLEEP_ON_IDLE_RUN` should be different.

> As for the OOP example,
Do you mean my own example for the OOP does not work? What are you running it on? I only tested on UNO I think.

GitMoDu, 2018-07-27:
> You can use my example 6 to validate without power consumption measurement.

Excellent, I will, thanks.

> Do you mean my own example for the OOP does not work?

Yes, example 21. I'm running on a STM32F1 (a.k.a. Mapple Mini, BluePill) with arduinoSTM32.

arkhipenko, 2018-07-27:
> Yes, example 21. I'm running on a STM32F1 (a.k.a. Mapple Mini, BluePill) with arduinoSTM32.

I don't know what to do! I don't have STM32F1, so can't test. Could it be the Arduino implementation on STM? Does it generate any additional messages you can send to me? 

`The most frequent causes of segmentation faults are generally pointer related: dereferencing an uninitialized pointer, a null pointer, or a previously freed pointer; accessing beyond the end (or in front of the beginning, but that's less frequent) of an object (array or other); using the results of an illegal pointer`

A null pointer should have segfaulted on other platforms, however, it is not uncommon for MC's to say nothing and just silently die. 
It is possible that creation of the Calculator object fails for some reason, and then pointer is null. Could you print the address of the Calculator here: after it is created? 

https://github.com/arkhipenko/TaskScheduler/blob/testing/examples/Scheduler_example21_OO_Callbacks/Scheduler_example21_OO_Callbacks.ino#L56

GitMoDu, 2018-07-30:
> I don't know what to do! I don't have STM32F1, so can't test.

You don't have to, but they cost 1.5€ with shipping included (throw in a few more cents and they'll come with the mapple mini bootloader already flashed, so you can use it with just a usb cable ), so there's really no reason not to get one 👍 
Otherwise, I'l probably use the hell out of your scheduler in near future, so I'll be testing and fixing things, if I can.

> It is possible that creation of the Calculator object fails for some reason, and then pointer is null.

I agree, it's most likely. 

Meanwhile I realized I am dumb: I cannot test sleep with a USB Serial being emulated by the host CPU, obviously. I'll try with a external serial adapter to see if I can test it properly. But you were right, the difference is visible with the Example 6, tried it with an AVR for reference.

arkhipenko, 2018-10-26:
@GitMoDu 
> You don't have to, but they cost 1.5€ with shipping included (throw in a few more cents and they'll come with the mapple mini bootloader already flashed, so you can use it with just a usb cable ), so there's really no reason not to get one
Do you have a link to the one you use with a bootloader? There are so many varieties - I want to test with your exact model

GitMoDu, 2018-10-26:
On ebay, just search for [Maple Mini STM32. ](https://www.ebay.com/itm/Mini-USB-STM32F103RCBT6-ARM-Cortex-M3-leaflabs-Leaf-maple-mini-module-F-arduino/312276765054?epid=1049477414&hash=item48b525497e:g:oLcAAOSwBRVbzpq6:rk:1:pf:0)

I have bought a few dozen, all came with bootloader.

I haven't forgotten this issue, I will report when I get the chance to do some testing.

arkhipenko, 2018-10-26:
> On ebay, just search for Maple Mini STM32.

Thank you! I just bought 3. Will play with it when it comes.

GitMoDu, 2018-11-11:
> Start without idle sleep.
> c1=4634635
> c2=1001
> Start with idle sleep.
> c1=4407173
> c2=1001

Ok, this is the best I got with sleep. I believe USB would have to be disabled for optimal results.

GitMoDu, 2018-11-12:
You can check out my attempts at https://github.com/GitMoDu/TaskScheduler/commits/testing

GitMoDu, 2018-11-13:
I am unable to test right now, but isn't that related to the other issue regarding constructors? I say this because my tasks are explicitly constructed and I never ran into this issue. Ex: MyTask(*Scheduler): Task( 0, TASK_FOREVER, scheduler, true).

arkhipenko, 2018-11-13:
>  I say this because my tasks are explicitly constructed and I never ran into this issue. Ex: MyTask(*Scheduler): Task( 0, TASK_FOREVER, scheduler, true).

I don't think so! I am checking the construction process - the task is constructed. But when the enable explicitly calls `OnEnable()` method, which is a virtual function, and should call the overloaded one, the whole thing hangs. Just hangs. On esp8266 it throws Exception(2), which according to this:
https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html
is `Processor internal physical address or data error during instruction fetch`
So my theory is it throws execution pointer to some incorrect address and it tries to execute random bytes as processor instructions. I just don't understand why....

GitMoDu, 2018-11-14:
When connected to USB, do you find a DFU USB device listed in your computer? If so, ignore the fact the current program has no serial (which is really used in the upload to reset the board), and press the reset button when you're uploading.
If not, and you find an unknown device, then you need to install the drivers.
If nothing shows up, you need to burn the bootloader at least once using a ST-Link.

arkhipenko, 2018-11-15:
Pushed to testing

closed 2018-11-15 by arkhipenko

arkhipenko, 2019-03-13:
> Also, try uploading using the Maple Mini board selected, instead of the generic STM32.
Got the board working and LOVING IT so far. Thanks for the tip!

arkhipenko, 2019-04-30:
> On ebay, just search for [Maple Mini STM32. ](https://www.ebay.com/itm/Mini-USB-STM32F103RCBT6-ARM-Cortex-M3-leaflabs-Leaf-maple-mini-module-F-arduino/312276765054?epid=1049477414&hash=item48b525497e:g:oLcAAOSwBRVbzpq6:rk:1:pf:0)

These have become my favorite MCs! I have bought 3, already burnt 2 and ordered 4 more!
Thanks again for the tip. They are awesome!

GitMoDu, 2019-05-07:
I see the bug has bitten you too. Speaking of, I'm gonna order another box of them :p

arkhipenko, 2019-05-07:
But Maple Labs discontinued those, and even the forum is closing down... What's the next thing?

chrisalbertson, 2019-05-07:
Yes, the "Maple" has been gone for a while.   The "Blue Pill" has pretty
much replaced it and it costs about half as much.
These things cost $2 each and have pretty much taken over the market.
ebay.com/itm/STM32-ARM-Tafel-STM32F103C8T6-ARM-Minimum-System
<https://www.ebay.com/itm/STM32-ARM-Tafel-STM32F103C8T6-ARM-Minimum-System-Entwicklung-Tafel-Module-BBC/143225408524?hash=item2158e64c0c:g:NToAAOSwl-FZjqUX>

The search terms are either "stm32 minimum system" or you can try "blue
pill" (the name is something to do with the Matrix movie)
This board is very well supported by the Arduino IDE now.

arkhipenko, 2019-05-07:
> Yes, the "Maple" has been gone for a while. The "Blue Pill" has pretty much replaced it and it costs about half as much. These things cost $2 each and have pretty much taken over the market. ebay.com/itm/STM32-ARM-Tafel-STM32F103C8T6-ARM-Minimum-System <https://www.ebay.com/itm/STM32-ARM-Tafel-STM32F103C8T6-ARM-Minimum-System-Entwicklung-Tafel-Module-BBC/143225408524?hash=item2158e64c0c:g:NToAAOSwl-FZjqUX> The search terms are either "stm32 minimum system" or you can try "blue pill" (the name is something to do with the Matrix movie) This board is very well supported by the Arduino IDE now.
> […](#)
> -- Chris Albertson Redondo Beach, California

Thank you. I saw them. Don't remember what stopped me - I should order a few and test with TaskScheduler just to make sure it works on this as well...

arkhipenko, 2019-05-07:
`Don't remember what stopped me`

Ah! This thing did not have a bootloader initially - but you can burn one, while Mini comes with it already.