arkhipenko/TaskScheduler #54: Cannot use <TaskScheduler.h> in several .h-files in one Project - Preprocessor-Statement issue?

PeiPei15, 2018-05-14:
I'm running into problems using this library in several headerfiles in my Project and I wonder if this is a problem with the preprocessor-statement because I'm not having this issue with other libraries like <vector.h> or else.

I could make it work using <TaskScheduler.h> in my Project.ino and <TaskSchedulerDeclarations.h> in my other Header-Files that need to work with Tasks and so on.

This is the minimal source-code that reproduces the error: [https://gitlab.com/narjesmieke/project](https://gitlab.com/narjesmieke/project)
My problem, the error as well as my current workaround is described in the following Stackoverflow-Post: [https://stackoverflow.com/questions/50291528/preprocessing-for-taskscheduler-library-doesnt-work-multiple-definition-error](https://stackoverflow.com/questions/50291528/preprocessing-for-taskscheduler-library-doesnt-work-multiple-definition-error)

arkhipenko, 2018-05-14:
You need to do it as described in this example:
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example16_Multitab

Duplicate of #51

arkhipenko, 2018-05-14:
Hi,

You need to do it as described in this example:
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example16_Multitab

Please let me know if that solves your issue.

Regards,
Anatoli