arkhipenko/TaskScheduler #58: Get upcoming task

jdk27, 2018-07-20:
It would be a good idea to be able to find the upcoming task that will be executed given the current task

arkhipenko, 2018-07-20:
Hi Josh,

Every task has member pointers to the next and previous task in the evaluation chain, which are currently private, but I can make them available via methods.

However, those are NOT the tasks which will be invoked next!
It is practically impossible to say which task will be invoked next because tasks can be triggered by events via Status Request objects, they can be enabled by other tasks, pin interrupt service routines, etc. etc.
So if you are asking to predict which task will be active after this one - it is "mission impossible" unfortunately.

Please send me your use case in case I misunderstood your question though.

Best,
Anatoli

jdk27, 2018-07-20:
Unfortunately I'm trying to get the next task to be invoked. Would there be a way to loop through each task in the evaluation chain passing it to the untilNextIterationFunction() to determine the upcoming task?

arkhipenko, 2018-07-21:
It will only work in simple and straightforward cases. What is the problem you are trying to solve? Maybe there is another way.

jdk27, 2018-07-22:
I'm trying to get my device to sleep between tasks, but I am using the feather M0 so I can't use the idle functionality from the library. Instead I'm using the rtc standbymode and setting the alarm to trigger right before the next task.

GitMoDu, 2018-07-23:
Why don't you implement the idle functionality for your M0? It's not that complicated, add a define and to sleep for 1 ms. Then you'll still be sleeping 99% of the unused time, with the just the light overhead of waking, checking scheduler and going to sleep again.

arkhipenko, 2018-07-24:
Based on the use case I see how this information could be used:
I can envision a scheduler collecting all the differences between current time and next invocation time for each of the tasks in the chain, and in the event when none of the callback methods was invoked, which means there is no way any of the past tasks' execution parameters were changed, the lowes of those values will be the time between now and the next task to be invoked. Which is a time interval the MC could be put to sleep. Any tasks starting on a pin or other hardware interrupt would wake up the MC, and as long as millis() continued to tick, scheduling should continue. 

I need to think how to implement this in a safe manner. I hate to touch the main scheduling code, as it is the core of the library and has been heavily optimized for speed. Really don't want to add cycles to it unless absolutely necessary.

arkhipenko, 2018-07-25:
Completely agree with @GitMoDu, and even more so, I really have to add anything to the scheduling loop that is not widely used but adds overhead to each evaluation pass. The goal is to minimize overhead as much as possible.

closed 2018-10-26 by arkhipenko

skezo, 2018-12-04:
@arkhipenko 

Would you consider making the pointers to next and previous task publicly available? This would allow us to loop through all current tasks in the scheduler. For example to find out what task IDs are currently set.

arkhipenko, 2018-12-05:
Yes. And I wanted to look at the Scheduler APIs and see what else could be required to manage the task chains.