arkhipenko/TaskScheduler #60: Compilation Error with PlatformIO for ESP8266

YannikW, 2018-08-08:
Hi,

I can't compile an empty PlatformIO project for ESP-8266 if TaskScheduler is importet.

Here is my `platormio.ini` File

```
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino

lib_deps =
    TaskScheduler
```

I got the following Error message:

```
Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=721
Downloading...
Unpacking  [########################------------]   67%
Unpacking...
Error: Traceback (most recent call last):
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 205, in process
    result = self._run()
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 300, in _run
    self.verbose)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 323, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 354, in install
    force=force)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 720, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 213, in _install_from_piorepo
    if app.get_setting("enable_ssl") else dl_data['url'], requirements)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 444, in _install_from_url
    self.unpack(dlpath, tmp_dir)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 224, in unpack
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 112, in unpack
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\yannik\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Users\Yannik\.platformio\python27\Lib\tarfile.py", line 2118, in extract
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name))
  File "C:\Users\Yannik\.platformio\python27\Lib\tarfile.py", line 2194, in _extract_member
    self.makefile(tarinfo, targetpath)
  File "C:\Users\Yannik\.platformio\python27\Lib\tarfile.py", line 2234, in makefile
    with bltn_open(targetpath, "wb") as target:
IOError: [Errno 2] No such file or directory: 'D:\\Eigene Dateien (cloud)\\GitLab\\LTC-Led-Track-Controller\\Software\\LTC-Led-Track-Controller\\.piolibdeps\\_tmp_installing-y9ze9y-package\\.\\examples\\Scheduler_example20_StatusRequest_LTS_WDT_Timeout_Object\\Scheduler_example20_StatusRequest_LTS_WDT_Timeout_Object.ino'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Any ideas what is causing the problem and how to solve it?

I reinstalled PlatformIO multiple times, tried Atom and VSCode, but everytime the same problem.

Thanks in advance.

arkhipenko, 2018-08-08:
Hi @YannikW 
Unfortunately, I am not using Platform.io at all, so not sure how to help you.

However, let's start with basic troubleshooting.
Are you able to compile the same code with any other 3rd party Arduino library?
(Trying to see if the problem is with TS or with Platform.io and how they handle ESP8266)

YannikW, 2018-08-09:
Ok, I tried something to find the issue..

First I tested to use TaskScheduler 2.6.0 instead of 2.6.1 using following platformio.ini file:

```
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino

lib_deps =
    TaskScheduler@2.6.0
```

And voila - it compiles without error!

But that can't be the solution.. So I tried something else:

My current project folder is `D:\Eigene Dateien (cloud)\GitLab\LTC-Led-Track-Controller\Software\LTC-Led-Track-Controller`

Then I tried to move the project folder to `C:\LTC-Led-Track-Controller`

And wow - 2.6.1 and 2.6.0 compiles without any problem..

So I guess the issue is not direct related to TaskScheduler..

Maybe there is an issue with the brackets or whitespaces in my project path which causes a weird error in combination with PlatformIO and TaskScheduler library

So for me that's closed. I don't know exactly what caused the problem, but I've found a way to solve it :D

But thanks anyway @arkhipenko for your reply

closed 2018-08-09 by YannikW

GitMoDu, 2018-08-09:
I'd advise you to raise the issue with Platform.IO guys, as this is clearly some old linux contributor who never uses spaces in files (because the web is also old and can't deal with spaces).