arkhipenko/TaskScheduler #61: Is running?

dragoblaztr, 2018-08-17:
is there a method or so to know if any task is running?

Best regards

arkhipenko, 2018-08-17:
TaskScheduler, being a cooperative scheduler, is inherently single-threaded. Therefore, only one task is truly running at a time. 
In other words: if a task1 is able to query the status of task2, then task1 is active (running) and task2 is inactive (not running).  It is always like this, so no additional method is needed.

dragoblaztr, 2018-08-17:
So is there any method to know what task are scheduled for running?

dragoblaztr, 2018-08-17:
My case is I want to stop and delete for schedule any specific task but it depends to be running on specific condition it whatever be running and existing or not? am I so specific?

arkhipenko, 2018-08-17:
Could you please describe your specific use case? The question is very broad. All enabled tasks on a chain are scheduled to run either on an event or after a specific time interval.

dragoblaztr, 2018-08-17:
I'm doing and alarm, when are enabled a green led on a panel are switched on, when a switch is opened alarm are triggered and a relay and red led (blinking) also in a same panel are switched on instead of green, and if are disabled if alarm was triggered previously turns off alarm and blinking red led, and make it turning on (without blinking), else just turns on red led and switch off the green one.

Thanks in advance

dragoblaztr, 2018-08-24:
I was digging in the source files and I found isEnabled(), can you explain how to this work because I looked at examples without luck.

Thanks

closed 2018-08-24 by dragoblaztr

arkhipenko, 2018-08-25:
This is method is explained in the wiki.
Could you please check there first?