arkhipenko/TaskScheduler #62: Power save/idle mode not working longer term?

David4x, 2018-08-30:
Firstly well done putting TaskScheduler together, looks powerful.

I tried your example 1 and added definition for task sleep on idle to confirm lower power consumption was possible.

It appears to be switching between normal power and power save initially but after task 2 ran a few times after task 3 was disabled it appears to never switch to idle and power is never reduced. I even increased task 2 to 5 seconds which should have made this even clearer.

I also tried example 18 but this does seem to work. 

Perhaps the bug is related to finite duration tasks, task disable or deleteTask?

GitMoDu, 2018-08-30:
Can you post an example code to replicate that behaviour? Seems odd.

David4x, 2018-08-30:
It was just example 1 but sorry false alarm I realised I placed the definition after the include rather than before as it should have been... 

`#define _TASK_SLEEP_ON_IDLE_RUN
#include <TaskScheduler.h>`

NOT

`#include <TaskScheduler.h>
#define _TASK_SLEEP_ON_IDLE_RUN`

closed 2018-08-30 by David4x

arkhipenko, 2018-08-30:
I love self-resolving issues!

:))))

Closed #62<https://github.com/arkhipenko/TaskScheduler/issues/62>.