arkhipenko/TaskScheduler #63: Rename enable() and enableDelayed()

GaryBoone, 2018-09-30:
The TaskScheduler is well-designed in allowing both immediate execution of tasks via *enable()* or delayed execution via *enableDelayed()*. The default however should be *enableDelayed()*, however, for a a few reasons:

1) running tasks immediately upon *enable()* means that the may execute at incorrect times. For example, a task may be set to run once per second, but if the *enable()* is called more than once in a second, the task will run more than once per second. This semantics means that the default behavior can lead to bugs. Programmers have additional burden to code this correctly.

2) "Enable" commonly means to make possible, not to do. So the function name is misleading.

3) For the same reason, enable is a function that should be idempotent. Intuitively, if you enable something that is already enabled, nothing should happen.

4) It doesn't scale. Suppose we have a thousand Tasks all with different timings. Suppose further that enabling them all via with *enableDelayed()* works: they all run correctly at their appointed times. But with the current semantics, using *enable()* and all of them would make them all run immediately, likely leading to conflicts or overload. 

Therefore *enable()* should use the delay, and *enableImmediate()* should be created to run immediately. Or to minimize disruption for existing users, deprecate *enable()*, leaving *enableImmediate()* and *enableDelayed()* to maximize explicitness and clarity.

GitMoDu, 2018-10-02:
My 2 cents on the issue.

#1 The previous [scheduler I was using](https://github.com/wizard97/ArduinoProcessScheduler/) conforms to enable = enableIfNot.
#2 As far as language is concerned, GaryBoone is correct that in English, enabling is explicitly not the same as activating. You enable something so you can can activate it.
#3 Completely understandable, if you value backwards compatibility, then this is a no-go.

arkhipenko, 2018-10-03:
And yet, according to Merriam-Webster

enable:
to cause (a feature or capability of a computer) to be active or available for use

https://www.merriam-webster.com/dictionary/enable

Anyway, backward-compatibility kind of seals the deal. I have literally thousands of lines of TS-based code that assumes immediate execution. Afraid I can't change  `enable` so radically.

closed 2018-10-04 by arkhipenko