arkhipenko/TaskScheduler #64: Real sensor delay

dragoblaztr, 2018-10-07:
I'm looking at example 5, and I have a question about how do I get the real sensor delay instead a simulated one like in this example?

dragoblaztr, 2018-10-08:
I'm using MAX6675 thermpcouple, how can I configure that sketch to measure that sensor timeout, since I noticed than in that example is like this:

> tSensor2.setInterval( random(1200) );

Thanks in advance

arkhipenko, 2018-10-08:
Example 5 simulates the behaviour of an ultrasonic sensor, which is measuring the time of sound wave travel, which, depending on the distance, is variable. Hence use of random() in the example. 

Thermocouple does not have a delay, and as far I was able to find out here:
[datasheet](https://datasheets.maximintegrated.com/en/ds/MAX6675.pdf) uses serial interface to send the measurement back to sketch (page 5). So there is no delay. 
You just use this library [adafruit](https://github.com/adafruit/MAX6675-library/blob/master/examples/serialthermocouple/serialthermocouple.pde) and read the value. 

I am not even sure you need to use code from example 5. Example 5 is a case of triggering events based on a number of sensors with unpredictable completion time. Does not seem like this is the case in your example.

dragoblaztr, 2018-10-08:
Ok, I'll take it on mind, thanks for the advice.

Issue closed

closed 2018-10-08 by dragoblaztr