arkhipenko/TaskScheduler #65: Add a default constructor to the Task class

CedricLor, 2018-10-13:
Hi,

It might be a good idea to add an explicit default constructor to the Task class.

Line 128 of TaskSchedulerDeclarations.h, add:
`INLINE Task() {};`

This would allow to compile libraries such as painlessMesh (which declare a bunch of Tasks without initializing them) together with option _TASK_STATUS_REQUEST.

Another question: why not use the classical header (.h) and code (.cpp) file structure instead of two .h files with different names?

arkhipenko, 2018-10-13:
It shod already be possible to instantiate Task() without any parameters. No change is required for that. 

No .CPP file is unfortunately the only way to make compilation parameters like _TASK_STATUS_REQUEST and others work with Arduino IDE.

CedricLor, 2018-10-13:
Thanks for your answer.

It was not possible if compiled with _TASK_STATUS_REQUEST

Accordingly, I forked your lib and modified both existing constructors:
- moved the taskScheduler param first,
undefaulted it 
and then added an explicit default constructor. Works fine this way for me anyway.

For the .CPP and conflict with Arduino IDE, I didn't know. Thanks;)

arkhipenko, 2018-10-13:
Oh, that's a bug then. Thanks for letting me know. Will fix!

arkhipenko, 2018-11-11:
Unfortunately, your solution changed the Task constructor signature and would make TS backwards-incompatible. 
I was able to resolve ambiguity just by disabling default values for Status Request constructor. Seems to work. 
Since Status Request enabled Tasks are a special case, it is ok for them to be explicitly constructed.
It also didn't seem to break any of my code and examples.

Will push into `testing` branch after a bit of testing and release together with 3.X

closed 2018-11-15 by arkhipenko