arkhipenko/TaskScheduler #66: OLED turn off after some time

tstanescu1, 2018-10-26:
Thank you for this great scheduler.. I've been able to do pretty much everything I've needed except one weird task.

I'm trying to set up a timeout on an OLED screen connected to the arduino. There is a button that can turn it off or on, but I'd like to implement a timeout after let's say 60 seconds to save the screen. 

I've set up all the tasks, and tried to use the t1.delay(60000); method but it didn't do anything, eg:

void displayWake() {
  
 u8x8.setPowerSave(false); //screen is on
 sleepTimerTask.setCallback(&displaySleep); //method sets power save to true
 sleepTimerTask.delay(60000); //delay before the function above is executed

}
 where the sleep timer is: Task sleepTimerTask(TASK_SECOND, TASK_ONCE, &displaySleep);

Thanks again for your help
T

arkhipenko, 2018-10-26:
Each task can be assigned a timeout since version 2.6.0
What you need to do is:

``` c++
#define _TASK_TIMEOUT   
#include <TaskScheduler.h>
...
void displaySleep();

Scheduler ts;
Task sleepTimerTask(TASK_MINUTE, TASK_ONCE, &displaySleep, ts, false);

void displaySleep() {
// do whatever needs to be done to turn the OLED Off
}

void displayWake() { 
  u8x8.setPowerSave(false); //screen is on
  sleepTimerTask.restartDelayed(); //restart the timer function
}
```
If anywhere in the code you want to reset the timeout (i.e., a button is pressed):

```
if (buttonPin == HIGH) {
  // do button stuff
  sleepTimerTask.restartDelayed(); //restart the timer function
}
```

tstanescu1, 2018-10-26:
Thanks so much for your reply. Unfortunately I can't seem to wrap my head around this timeout. I've tried several combinations and it doesn't seem like the displaySleep function is engaging, it doesn't sleep.  Do I need to enable anything else? What about using something like the following, where I could add one task after the other?

Task t2(5 * TASK_SECOND, TASK_FOREVER, &task2Callback, &ts, false, NULL, &task2OnDisable);

tstanescu1, 2018-10-26:
Here is the code I have right now, it's quite messy sorry:
`//Temperature
#include <OneWire.h> //Used for temp
int DS18S20_Pin = 4; //DS18S20 Signal pin on digital 2
OneWire ds(DS18S20_Pin);  //Temperature chip i/o

//TDS Sensor
#define TdsSensorPin A1
#define VREF 5.0      // analog reference voltage(Volt) of the ADC
#define SCOUNT  30           // sum of sample point
int analogBuffer[SCOUNT];    // store the analog value in the array, read from ADC
int analogBufferTemp[SCOUNT];
int analogBufferIndex = 0, copyIndex = 0;
float averageVoltage = 0, tdsValue = 0;

//PH Sensor
#define SensorPin A0            //pH meter Analog output to Arduino Analog Input 0
#define Offset 0.00            //deviation compensate
#define samplingInterval 20
#define printInterval 800
#define ArrayLenth  40    //times of collection
int pHArray[ArrayLenth];   //Store the average value of the sensor feedback
int pHArrayIndex = 0;

/* Simple version of the wake up on capacitive button */
#include <U8x8lib.h> //Display library
#include <TaskScheduler.h>

//Capacitative Button
int pushButton = 11;

#define DISPLAY_AUTOSLEEP_MS 15000  // sleep display on idle (after ms)
#define DISPLAY_BLIT_FREQ 1000       // redraw display every second

// Forward declarations for Display
void blitDisplay(); //Function that draws the displays data on OLED, but no backlight here
void tick(); //Function that senses rotary click or movement to wake up screen
void displaySleep(); //Function that makes screen sleep
void displayWake(); //Function that wakes up screen

//void ph();

// logic variables
//int timelapseInterval = 0;
//bool timelapseRunning = false;
bool blit = true;

// Library initializations
U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);

Scheduler scheduler;
Task sleepTimerTask(TASK_SECOND * 10, TASK_ONCE, &displaySleep, &scheduler, false);
Task blitDisplayTask(DISPLAY_BLIT_FREQ, TASK_FOREVER, &blitDisplay);

//Task sensorDataSend(TASK_SECOND, TASK_FOREVER, &sensorRead);

void setup(void) {

  pinMode(pushButton, INPUT); //Capacitive touch sensor
  pinMode(TdsSensorPin, INPUT); //TDS/PPM Sensor
  u8x8.begin();
  //u8x8.setFont(u8x8_font_pxplusibmcgathin_r);
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.setContrast(0); //dims it down

  // Initialize the scheduler.
  scheduler.init();
  scheduler.addTask(blitDisplayTask);
//  scheduler.addTask(sleepTimerTask);
  //blitDisplayTask.enable();
[... 238 lines cut ...]
//  }

  ds.reset();
  ds.select(addr);
  ds.write(0x44, 1); // start conversion, with parasite power on at the end

  byte present = ds.reset();
  ds.select(addr);
  ds.write(0xBE); // Read Scratchpad

  for (int i = 0; i < 9; i++) { // we need 9 bytes
    data[i] = ds.read();
  }

  ds.reset_search();

  byte MSB = data[1];
  byte LSB = data[0];

  float tempRead = ((MSB << 8) | LSB); //using two's compliment
  float TemperatureSum = tempRead / 16;

  return TemperatureSum;

}`

arkhipenko, 2018-10-26:
Let's take this offline from the issues. Please send me an email to arkhipenko@hotmail.com

closed 2018-10-26 by arkhipenko

arkhipenko, 2018-10-30:
@tstanescu1 : Do you still want the updated code? I have it written just need your email to send it to.
Please send me an email to arkhipenko@hotmail.com

tstanescu1, 2018-11-01:
Hi, i'm sorry I had to pause my project for a bit. I will send you an email as soon as I start it back up.

Thank you for all your help!