arkhipenko/TaskScheduler #67: Add->Delete->Add task doesn't start task

vadimostanin, 2018-11-09:
//Task t2 isn't launched in t1

#include <TaskScheduler.h>

// Callback methods prototypes
void t1Callback();void t2Callback();

//Tasks
Task t1(2000, 30, &t1Callback);
Task t2(5000, TASK_FOREVER, &t2Callback);

Scheduler runner;

void t1Callback() {
    Serial.print("t1: ");
    Serial.println(millis());
    if (t1.isLastIteration())  {
      runner.addTask(t2);
      t2.enable();
      t2.disable();
      runner.deleteTask(t2);
      runner.addTask(t2);
      t2.enable();
      Serial.println("t1: start, stop, start, t2");
    }
}

void t2Callback() {
    Serial.print("t2: ");
    Serial.println(millis());
  
}

void setup () {
  Serial.begin(9600);
  Serial.println("Scheduler TEST");
  
  runner.init();
  Serial.println("Initialized scheduler");

  runner.addTask(t1);
  Serial.println("added t1");
  
  t1.enable();
  Serial.println("Enabled t1");
  t2.enable();
  Serial.println("Enabled t2");
}

void loop () {
  runner.execute();
}

arkhipenko, 2018-11-09:
It's a bug. 
I guess I have never used and therefore tested deleting a task from the chain and then adding it back.

``` c++
runner.addTask(t2);
t2.enable();
t2.disable();
runner.deleteTask(t2);
runner.addTask(t2);
t2.enable();
```
Will fix in the testing branch in the nearest and release with 3.0

arkhipenko, 2018-11-09:
Pushed into `testing` branch. Will be released with major 3.X release.

closed 2018-11-09 by arkhipenko