arkhipenko/TaskScheduler #68: No Task default contructor if _TASK_STATUS_REQUEST

vadimostanin, 2018-11-09:
I defeined _TASK_STATUS_REQUEST, then I can't create empty Task object, because there is
two constructors that are satisfy conditions with empty arguments:
    INLINE Task(unsigned long aInterval=0, long aIterations=0, TaskCallback aCallback=NULL, Scheduler* aScheduler=NULL, bool aEnable=false, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);

#ifdef _TASK_STATUS_REQUEST
    INLINE Task(TaskCallback aCallback=NULL, Scheduler* aScheduler=NULL, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);

arkhipenko, 2018-11-09:
Yes, it's a known problem - I just don't have time to fix it.
Workaround: do not use default Task t; constructor. 
Use something like Task t(&callback, &scheduler);

vadimostanin, 2018-11-09:
Yes, seems acceptable workarounds exists. I will close issue.

closed 2018-11-09 by vadimostanin