arkhipenko/TaskScheduler #69: Add setInterval() signature that supports std::chrono

moritz89, 2018-12-18:
The current setIteration() function does not type check the duration that is passed to it. It is convention that the int represents milliseconds (unless microsecond support is enabled). It would be safer if it were possible to use std::chrono types such as

- std::chrono::seconds
- std::chrono::milliseconds
- std::chrono::microseconds

Would this extension be within the project's philosophy?

GitMoDu, 2018-12-18:
The standard library is mostly incompatible and is not widely used with MCUs. Also, the, overhead is not insignificant (32 bit millis becomes 45 bits), bearing in mind that for a 8 bit MCU to munch 32 bit numbers is already slow.

arkhipenko, 2018-12-18:
I agree with @GitMoDu: the std library will not compile on Arduinos (that is why it is only supported for ESP8266's), and additionally I don't really see the point in type checking for a library that only gives you two mutually exclusive choices: either `millis` or `micros`, not both. 
The programmer needs to decide once and then whatever you pass to `setInterval`, task constructor or just `set` method in terms of timing is always that - either millis or micros. There should not be any doubt or confusion about what the parameter represents.

moritz89, 2018-12-18:
Makes sense that it would be adding a heavy dependency. Either way, thanks for a great tasking library

closed 2018-12-18 by moritz89