arkhipenko/TaskScheduler #70: A quick question (disable, delete task from callback)

m-menon, 2018-12-25:
Pretty cool library, thanks! A simple question - is it ok if I disable and delete a task from it's callback? Let me explain - there is a task that keeps running until a certain condition is met. The callback of the task keeps checking for the condition. Once the condition is met, the task is disabled and deleted ( from the same task's callback ). Is this ok? The reason I ask is because I am deleting the task from within it's own callback. Just wondering if that's bad or would that cause any weird behavior or some unexpected results. It seems to work but I thought I'd ask. Thanks a lot.

arkhipenko, 2018-12-25:
It is ok for sure to delete a task from its OnDisable method - I tested that.

As far as Callback is concerned, should be ok as long as there is no follow up references, like OnDisable called right after. Generally it is a good practice to delete and return immediately.

Please do test!

arkhipenko, 2019-03-26:
A word of caution: apparently the ability to delete "itself" in the OnDisable method depends on the toolchain. On STM32F1 this does not work, while on Arduino and ESP8266 it does. Go figure!

GitMoDu, 2019-03-26:
That's very odd. I'll try it on STM32F1.

arkhipenko, 2019-03-27:
> That's very odd. I'll try it on STM32F1.

Try running example 19 (minus the `FreeMem()` ) on STM32F1. In my case, the Tasks are not properly deleted (the total # of tasks never goes to 0 reliably). In another case the sketch hangs in the OnDelete() method right at `delete t;` line with the blue LED blinking rapidly (as if a null pointer reference threw it into the bootloader?). I could not find an explanation of what a fast blinking blue LED means on Mini...

GitMoDu, 2019-03-29:
Fast blinking blue LED means seg-fault a.k.a. null pointer exception.