arkhipenko/TaskScheduler #74: Conflict when used with other library

m-menon, 2019-04-27:
When I use this library along with BLE library, there seems to be some conflict. Independently both the BLE and the taskScheduler libraries work great but when I put them together ( even in the simplest of sketches ), the following error comes up

```
...\arduino_build_304058\libraries\ESP32_ble_library\Task.cpp.o: In function `Task::~Task()':

...\Arduino\libraries\ESP32_ble_library/Task.cpp:36: multiple definition of `Task::~Task()'

...\arduino_build_304058\sketch\BLE_scan_new_lib.ino.cpp.o:...\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:255: first defined here

...\arduino_build_304058\libraries\ESP32_ble_library\Task.cpp.o: In function `Task::~Task()':

Task.cpp:(.text._ZN4TaskD2Ev+0x0): multiple definition of `Task::~Task()'

...\arduino_build_304058\sketch\BLE_scan_new_lib.ino.cpp.o:BLE_scan_new_lib.ino.cpp:(.text._ZN4TaskD2Ev+0x0): first defined here

collect2.exe: error: ld returned 1 exit status
```

I tried a few things and I am still at it. Perhaps this is a known issue or something?

m-menon, 2019-04-28:
Oops, I just tried adding this line right before the TaskScheduler.h include

```
#define _TASK_INLINE // Make all methods "inline" - needed to support some multi-tab, multi-file implementations
#include <TaskScheduler.h>
```

And the error's gone! I was thinking of trying out the other method - splitting stuff up into different files. I guess now all is fine. Thanks for the great library and especially the `#define _TASK_INLINE`

closed 2019-04-28 by m-menon

m-menon, 2019-04-28:
@arkhipenko Btw I just wanted to say thanks. This lib is very well written and I am sure it's come handy to many ppl on many occasions. As a small token of appreciation I'd want to get you couple of beers if that's ok. Do you have a PayPal donate button anywhere? :)

GitMoDu, 2019-04-29:
#include <TaskScheduler.h> should only be declared on the main.cpp.
For every other use class, you must use #include <TaskSchedulerDeclarations.h>

m-menon, 2019-04-29:
Thanks GitMoDu. I'll try that out. The `#define _TASK_INLINE` seems to have solved the problem for now.

GitMoDu, 2019-04-29:
You're probably duplicating a lot of ROM code because of that, but I might be wrong. Nothing like testing :)

arkhipenko, 2019-04-29:
> Do you have a PayPal donate button anywhere? :)

Hi, sorry for late reply - been busy. 
No worries about the beer - I am just sharing with the community. But thank you for the gensture - appreciate it. 

If your project is multi-tab (assuming you are using Arduino IDE) then yes, what @GitMoDu pointed out should solve the problem. 
I don't think you'd want to use FreeRTOS and TaskScheduler in the same project. It looks like ESP32 BLE uses it internally and the conflict is somewhere with the declaration of Tasks... I have not looked further as I am not developing on ESP32 actively at the moment. Focusing on STM32F103RCBT6 Maple Mini boards per suggestion of the same @GitMoDu  :)))

arkhipenko, 2019-04-29:
> You're probably duplicating a lot of ROM code because of that, but I might be wrong. Nothing like testing :)

But it runs faster that way :)))))

m-menon, 2019-04-30:
hi @arkhipenko, thanks for your reply and clarifying. I did try things in the same way as in "Scheduler_example16_Multitab" but I did hit some issues and I left it at that. 

For now the `#define _TASK_INLINE` seems to work without issues and I haven't really made a comparison of sizes before and after. My sketch uses 50% of program storage space and Global variables use (19%) of dynamic memory. So I guess it ain't so bad :) I'll report any new findings as I experiment. I'll try @GitMoDu's suggestion as well. 

And thanks a lot - appreciate your help so much :)