arkhipenko/TaskScheduler #77: seems not to run with Teensy 4.0

webline, 2019-09-01:
no tasks get executed on a Teensy 4.0 with a Cortex-M7 processor. The same code works fine on my Teensy 3.2 or 3.6

arkhipenko, 2019-09-01:
No idea. I don't have Teensy 4. As you stated, it ran ok on T 3.5 for me. I need to read up about T4...

Could it be that it puts the chip to sleep permanently? Did you try without TASK_IDLE_SLEEP?

webline, 2019-09-01:
the fast Teensy 4 would be a great platform for a complex task driven system and I love your lib!
Your tip is great: if I remove

#define _TASK_SLEEP_ON_IDLE_RUN

the tasks get executed again - great!

GitMoDu, 2019-09-02:
Seems like someone needs to implement SLEEP_ON_IDLE for the new Teensy.

arkhipenko, 2019-09-02:
"Someone" needs to procure a new Teensy first... Unless seonebelse wants to give it a try...

arkhipenko, 2020-02-25:
Any updates on how to properly put Teensy 4 to sleep?

arkhipenko, 2020-03-16:
Closing this since I have no Teensy 4.0. Anyone wants to contribute - please feel free.

closed 2020-03-16 by arkhipenko