arkhipenko/TaskScheduler #79: add an understandable documentation and examples that really show how it works

StefanL38, 2019-10-08:
I have read your "simple" example. This example might run. But I understand nothing. It makes use of delay all the time. Is this a complete FAKE confusing newbees??? 

At least you should explain why in your code delay does NOT block code-execution

arkhipenko, 2019-10-09:
Since there is no link to which "simple" example you are talking about, I can not help you.
In fact, the tone of your comment is such that I really do not want to. 
You are welcome to NOT use this library.

closed 2019-10-09 by arkhipenko

StefanL38, 2019-10-09:
Hello,

I apologise for writing in a rude manner. I was looking at multiple libraries at the same time.
After relooking inside your examples I see that my comment was meant for another multitask library.
Again I apologise for what I have written. Completely my fault. In fact THIS library is well documented. 

You are right I am new to github. I was not aware that documentation is placed under subfolder "extras".

So I hope you accept my apologising and - after reading your documentation - if I have any further questions that you might reply on them

best regards  

Stefan