arkhipenko/TaskScheduler #84: extern millis declaration

BlackEdder, 2020-01-15:
I get the follow error with later versions of TaskScheduler/esp:

```
 In file included from lib/painlessMesh/src/painlessMeshConnection.h:6:0,
                 from lib/painlessMesh/src/painlessMeshConnection.cpp:9:
/root/.platformio/lib/TaskScheduler_ID721/src/TaskSchedulerDeclarations.h:48:22: error: previous declaration of 'long unsigned int millis()' with 'C++' linkage
 extern unsigned long millis(void);
                      ^
In file included from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:25:0,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
                 from lib/painlessMesh/src/painlessMeshConnection.h:11,
                 from lib/painlessMesh/src/painlessMeshConnection.cpp:9:
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:187:26: error: conflicts with new declaration with 'C' linkage
 unsigned long millis(void);
                          ^
In file included from lib/painlessMesh/src/painlessMeshConnection.h:6:0,
                 from lib/painlessMesh/src/painlessMeshConnection.cpp:9:
/root/.platformio/lib/TaskScheduler_ID721/src/TaskSchedulerDeclarations.h:47:22: error: previous declaration of 'long unsigned int micros()' with 'C++' linkage
 extern unsigned long micros(void);
                      ^
In file included from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:25:0,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
                 from lib/painlessMesh/src/painlessMeshConnection.h:11,
                 from lib/painlessMesh/src/painlessMeshConnection.cpp:9:
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:188:26: error: conflicts with new declaration with 'C' linkage
 unsigned long micros(void);
                          ^
*** [.pio/build/nodemcuv2/lib771/painlessMesh/painlessMeshConnection.cpp.o] Error 1
```

I assume this would be solved by wrapping them as follows in the TaskSchedulerDeclarations file (but have not tested this).

```
extern "C"
{
   unsigned long micros(void);
   unsigned long millis(void);
}
```

arkhipenko, 2020-01-16:
Would it also help if I moved this declaration into TaskScheduler.h file, which must only be included once?

TD-er, 2020-01-16:
declare it using extern in a .h file and implement it in only a single .cpp file.

Something like this...

`some.h` file:
```c++
extern unsigned long micros(void);
extern unsigned long millis(void);
```

`some.cpp` file:
```c++
#include "some.h"

unsigned long micros(void) { .... }
unsigned long millis(void) { .... }
```

This allows the .h file to be included as much as you like.
The same for (global) variables.
Declare as extern and construct in only a single .cpp file.

arkhipenko, 2020-01-16:
I can't have any .cpp files in the library due to the way Arduino IDE handles #define statements...

arkhipenko, 2020-01-17:
Could you please check if the version I just pushed into "testing" branch solves the issue? 
Thx

Silver-Void, 2020-01-17:
I had the same issue but the testing branch version solved the compilation errors for me.

arkhipenko, 2020-01-17:
Pushed into master as 3.1.2

closed 2020-01-17 by arkhipenko

BlackEdder, 2020-01-20:
Thank you this has now been solved.