arkhipenko/TaskScheduler #85: new version dont work STM32

nestorlevi, 2020-01-30:
hi. I was testing example 02, which used to work with the STM32F103C with the previous version, now generates the following error:

```
stm32duino\hardware\STM32F1\2020.1.24\cores\maple/wirish_time.h:54:22: error: previous declaration of 'uint32 micros()' with 'C++' linkage

 static inline uint32 micros(void) {

\TaskScheduler\src/TaskScheduler.h:167:30: error: conflicts with new declaration with 'C' linkage

     unsigned long micros(void);

stm32duino\hardware\STM32F1\2020.1.24\cores\maple/wirish_time.h:45:22: error: previous declaration of 'uint32 millis()' with 'C++' linkage

 static inline uint32 millis(void) {

TaskScheduler.h:168:30: error: conflicts with new declaration with 'C' linkage

     unsigned long millis(void);
```
It seems that there is a redefinition of functions, any solution?

arkhipenko, 2020-01-30:
It seems each board package defines `millis()` and `micros()` differently. 
I can introduce a `_TASK_DEFINE_MILLIS` compile option to force definition on the platforms that require it. 
Really don't want to account for every single board.

arkhipenko, 2020-01-30:
Just pushed an update into testing branch - please check. 
I was able to compile tests for various boards with the change.

nestorlevi, 2020-01-31:
It works correctly, thanks.

arkhipenko, 2020-01-31:
Will push into master shortly

arkhipenko, 2020-02-17:
Merged into 3.1.3
Use _TASK_DEFINE_MILLIS define to force the forward definition of millis() and micros()

closed 2020-02-17 by arkhipenko