arkhipenko/TaskScheduler #87: Please fix compiler warnings

Jingjok86, 2020-02-20:
Hi,
First let me say, thank you for the TaskScheduler library, really useful with powerful functionality.

As a minor issue, could you please though fix the compiler warnings about unused variables, it makes it more difficult to spot local project warnings. The warnings I get with the newest library version are :

TaskScheduler.h: In member function 'bool Scheduler::execute()':
TaskScheduler.h:904:19: warning: variable 'tStart' set but not used [-Wunused-but-set-variable]
     unsigned long tStart, tFinish;
TaskScheduler.h:904:27: warning: variable 'tFinish' set but not used [-Wunused-but-set-variable]
     unsigned long tStart, tFinish;
TaskScheduler.h:909:19: warning: unused variable 'tIdleStart' [-Wunused-variable]
     unsigned long tIdleStart = 0;

I have set only one define :
#define _TASK_TIMECRITICAL

arkhipenko, 2020-02-23:
Please test v3.1.4 from the testing branch.

Jingjok86, 2020-02-23:
Hi Anatoli,

still the same warnings in the code from the testing branch.

What fixes it is using only condition #ifdef _TASK_SLEEP_ON_IDLE_RUN 
instead of the condition
    #if defined (_TASK_SLEEP_ON_IDLE_RUN) || defined (_TASK_TIMECRITICAL)

i.e.
    #ifdef _TASK_SLEEP_ON_IDLE_RUN
        unsigned long tStart, tFinish;
    #endif
...
    #ifdef _TASK_SLEEP_ON_IDLE_RUN
        tStart = micros();  // Scheduling full pass start time in microseconds. 
    #endif
...
    #ifdef _TASK_SLEEP_ON_IDLE_RUN
        tFinish = micros(); // Scheduling pass end time in microseconds.
    #endif

because both variables are only used in line 1060, if _TASK_SLEEP_ON_IDLE_RUN is defined.

arkhipenko, 2020-02-23:
Hmm. Shooting in the dark here as I don't get those messages... Ok. Let me check again.

arkhipenko, 2020-02-23:
Please try now and retest the testing branch version 3.1.4

Jingjok86, 2020-02-24:
Hi,
OK, almost there, one warning about tIdleStart remaining, it can be fixed by changing line 924 and the next few from :

```
#ifdef _TASK_TIMECRITICAL
    register unsigned long tPassStart;
    register unsigned long tTaskStart, tTaskFinish;
    unsigned long tIdleStart = 0;
#endif  // _TASK_TIMECRITICAL
```

To :

```
#ifdef _TASK_TIMECRITICAL
    register unsigned long tPassStart;
    register unsigned long tTaskStart, tTaskFinish;
#endif 

#if defined (_TASK_SLEEP_ON_IDLE_RUN) && defined (_TASK_TIMECRITICAL)
    unsigned long tIdleStart = 0;
#endif
```

I am btw using ESP32 Arduino core on Visual Studio Code and Platform IO with default compiler warning settings, so the warnings should come up for all users with this fairly common setup.

arkhipenko, 2020-02-25:
Ok. Third time is a charm. Please try now.

Jingjok86, 2020-02-25:
Hi,
for my use case it is fine now, no compiler warnings anymore, I use only _TASK_TIMECRITICAL

But I would recommend to change line 928 from

```
#ifdef _TASK_SLEEP_ON_IDLE_RUN
```

to

```
#if defined (_TASK_SLEEP_ON_IDLE_RUN) && defined (_TASK_TIMECRITICAL)
```

as per my last post, because else users that use only the flag _TASK_SLEEP_ON_IDLE_RUN without _TASK_TIMECRITICAL will still get the warning about tIdleStart

anyway, you can close the issue, thanks !

GitMoDu, 2020-04-01:
> TaskSchedulerDeclarations.h:267: note  type struct Scheduler itself violates the C++ One Definition Rule.

This is new.

arkhipenko, 2020-04-01:
Line 267 is `#endif  // _TASK_LTS_POINTER`
???

arkhipenko, 2020-04-01:
It must be giving reference to the file resulting from a combination of several `#defines...` tho.
How can I get to your line count?  :))))

GitMoDu, 2020-04-05:
Just pulled the latest update, still get the same warning.
The line indicated is the declaration of Scheduler class.

> class Scheduler {
>   friend class Task;
>   public:

I'm using _TASK_OO_CALLBACKS and _TASK_SLEEP_ON_IDLE_RUN. If I remove the _TASK_SLEEP_ON_IDLE_RUN, then the warning doesn't appear.

arkhipenko, 2020-04-07:
What is wrong with this statement? And why did it become a warning suddenly?
I am not sure what I should do here.

GitMoDu, 2020-04-11:
> And why did it become a warning suddenly?

Probably because of Arduino's compiler version update, which is becoming more strict. I'll let you know what I find.

GitMoDu, 2020-04-24:
Ok, so PVS static analyser found nothing, but I'm still getting the warning.
I did find a [similar issue](https://stackoverflow.com/questions/40659435/one-definition-rule-warning), where in the accepted answer we can find:
> implicitly-defined inline constructor

So something to do with the inline constructors. Since this warning goes away if I disable _TASK_SLEEP_ON_IDLE_RUN, then where there's smoke there's fire.

arkhipenko, 2020-04-24:
Is it possible that this is caused by a name clash with another library? Afterall "Task" and such a common term.
I am asking because I am on the latest Arduino IDE and I do not get this warning.

MikeBlueIce, 2020-04-26:
I get  "multiple definition of iSleepMethod" and "multiple definition of iSleepScheduler"  with _TASK_SLEEP_ON_IDLE_RUN defined and multiple cpp files referencing to taskscheduler.

GitMoDu, 2020-04-27:
[Related.](https://github.com/jrowberg/i2cdevlib/issues/424)

Seems like it's Visual Micro (which I use) that's issuing that warning.

arkhipenko, 2020-05-04:
I installed Visual Micro and compiled a TS-based project with sketch and library warnings enabled - did not get this message.  Don't know what to do so I am going to close this issues.

closed 2020-05-04 by arkhipenko