arkhipenko/TaskScheduler #88: strange compiling errors

ilovesushi, 2020-02-21:
fresh vscode+platformio+stm32 arduino project
blink example cant be compilled
library added via platformio 
didnt compile at all if were copied to lib(folder)
what can it be

`#define _TASK_SLEEP_ON_IDLE_RUN
#include <TaskScheduler.h>
//#define  LEDPIN    13
int static LEDPIN=PC13;
Scheduler ts;

void WrapperCallback();

void BlinkOnEnable();
void BlinkOnDisable();
void LEDOn();
void LEDOff();

Task tWrapper(30000, TASK_FOREVER, &WrapperCallback, &ts, true);
Task tBlink(5000, TASK_ONCE, NULL, &ts, false, &BlinkOnEnable, &BlinkOnDisable);
Task tLED(TASK_IMMEDIATE, TASK_FOREVER, NULL, &ts, false, NULL, &LEDOff);

void WrapperCallback() 
{
    Serial.println("In WrapperCallback");
    tBlink.restartDelayed();
    // LED blinking  is initiated
    //every 30 seconds for 5 seconds 
}
{
    Serial.println("In WrapperCallback");
    tBlink.restartDelayed();
    // LED blinking  is initiated
    //every 30 seconds for 5 seconds 
}
`
`-- <TaskScheduler> 3.1.3
Building in release mode
Compiling .pio\build\blackpill_f103c8\src\main.cpp.o
src\main.cpp:17:48: error: invalid conversion from 'void (*)()' to 'TaskOnEnable {aka bool (*)()}' [-fpermissive]
 Task tBlink(5000, TASK_ONCE, NULL, &ts, false, &BlinkOnEnable, &BlinkOnDisable);
                                                ^~~~~~~~~~~~~~
In file included from src\main.cpp:3:0:
C:\Users\dan\.platformio\lib\TaskScheduler_ID721\src/TaskScheduler.h:247:1: note:   initializing argument 6 of 'Task::Task(long unsigned int, long int, TaskCallback, Scheduler*, bool, TaskOnEnable, TaskOnDisable)'
 Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {
 ^~~~
src\main.cpp:28:1: error: expected unqualified-id before '{' token
 {
 ^
src\main.cpp: In function 'bool BlinkOnEnable()':
src\main.cpp:38:6: error: ambiguating new declaration of 'bool BlinkOnEnable()'
 bool BlinkOnEnable()
      ^~~~~~~~~~~~~
src\main.cpp:11:6: note: old declaration 'void BlinkOnEnable()'
 void BlinkOnEnable();
      ^~~~~~~~~~~~~
*** [.pio\build\blackpill_f103c8\src\main.cpp.o] Error 1
=========================================== [FAILED] Took 3.78 seconds ===========================================`

ilovesushi, 2020-02-21:
``

closed 2020-02-21 by ilovesushi

arkhipenko, 2020-02-21:
Documentation was developed before the Arduino IDE started to require forward function definition.
So Wiki Blink example would not even compile in 1.8.10 - forward definitions need to be added. 
The `on enable` functions must be `bool`, not `void` - which you have already figured out. 
I wish I have the time to go back and review all the documentation, but I don't - sorry - when I figure out the way to monetize TS and live off of it, I promise, the documentation will be stellar!

ilovesushi, 2020-02-23:
I think you need more to do to get  living from it:)
By the way do you know any useful resources to help me out to rewrite i2c spi rs232 asynchronously with Tasks schedule r