arkhipenko/TaskScheduler #89: Cannot schedule tasks using member functions

Lerring, 2020-03-10:
Hi,

I can't seem to find any documentation on this online - Is it expected behaviour to be unable to assign tasks to non-static member functions?

I'm attempting to avoid using delay to turn a buzzer on and off as it messes up the timings of the rest of my scheduled tasks.

Kind regards

arkhipenko, 2020-03-10:
Are you trying to create a Task that turns buzzer on and off depending on the call to update?
If that is the intent:

1. I always use a static instance of scheduler object (one or several if you need layered prioritization). 
The reason is: the `scheduler `is a heartbeat of task execution, it lives above all tasks and objects and makes things run. Therefore I do not support hiding schedulers inside classes. 
All of the above is also the reason why my `loop()` is always one line: `ts. execute()`
(ts is a static scheduler object)

2. Example 15 is focused on the use of _standard functions_, which I think is a kind of lambda function in C++. This example was provided by a contributor. I do not personally use lambdas (nothing wrong with them - jut did not feel the need). So I can't help you there.

3. I would direct your attention to [example 21](https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example21_OO_Callbacks) for using TS in an object-oriented way. 

If I was writing your code I would separate the IObserver and actual Buzzer tasks. 
I would make Buzzer start buzzing in the `OnEnable()` method and Stop buzzing in the` OnDisable()` method. There is no need for the actual callback so that could remain empty. 
I would implement Buzzer derived from Task class with no additional public methods. 
You may want to create a private `start()` and `stop()` methods and call them from `OnEnable `and `OnDisble` respectively. 
Your IObserver-derived class then just needs to call `buzzer.enable()` when an enable event arrives and `buzzer.disable()` when a disable event arrives. 
A good reference is a SuperSensor class from example 21, but buzzer should be way simpler. 
Please do not forget to compile with `#define _TASK_OO_CALLBACKS` compile option.

That's it. 
Hope this helps.
 
Regards,
Anatoli

arkhipenko, 2020-03-10:
And to answer the title of the issue:
Cannot schedule tasks using member functions

By default only static callbacks are supported. 
To enable dynamic method binding you need to compile with `_TASK_OO_CALLBACKS` option and follow example 21. I do not have the Wiki page for OO method yet. I actually never felt the need to switch to it - the sketches are short enough to use traditional Arduino approach.

Lerring, 2020-03-10:
> Are you trying to create a Task that turns buzzer on and off depending on the call to update?

The intention of this is to buzz for X seconds without using a delay - this is because i can imagine it causes a block messing up the timings of my scheduled events which may want to run in this time frame. The event class is abstract and will contains an Integer which determines the length to buzz for.  

> I would make Buzzer start buzzing in the OnEnable() method and Stop buzzing in the OnDisable() method.

I was initially thinking about sending events which say ON / OFF, though that would move my problem to another class of needing a non blocking delay if they had to be run manually, If the OnDisable() is called automatically after the task has run its duration, making the buzzer a task would work I can imagine.  

I currently have a class which wants to enable the buzzer every 5 / 30 seconds (depending on room temperature) to alert for X seconds, the enabling works by sending the event containing X following the observer pattern. 

I'll have a look at example 21 and see if that can help clarity.

Thanks for your input, I really appreciate the thoughts

Kind regards,

Lerring

arkhipenko, 2020-03-10:
No worries. You could also send your full sketch to arkhipenko@hotmail.com - I can look over and make suggestions. 

Just to make sure - this is a cooperative multitasking library, not pre-emptive. You have to use TS for **all** your tasks, not just the buzzer. (Asking because you were using private scheduler as a member inside the task - that not the best way and won't work in many cases as scheduler does not have control outside of its task chain).

Lerring, 2020-03-10:
I've sent over an email.

TS is the only task scheduler I have used, One acting as a heart beat of the task execution and the second in this example.

arkhipenko, 2020-03-16:
Closing this thread as it moved to PM space.

closed 2020-03-16 by arkhipenko