arkhipenko/TaskScheduler #90: Online Playground for TaskScheduler

urish, 2020-03-13:
Hello, I came across this library, and really love it - finally, a more structured solution than using `millis()` everywhere!

I took the LED blinking example and created an online playground for it on Wokwi, an Arduino teaching platform I'm working on. The playground allows you to run the code inside your browser, using an [Arduino simulator](https://blog.wokwi.com/avr8js-simulate-arduino-in-javascript/?utm_source=github&utm_medium=social&utm_campaign=TaskScheduler), so you can learn how the API works and experiment with it right from your browser, without the need to pull and connect an Arduino board to your PC:

[https://wokwi.com/playground/task-scheduler](https://wokwi.com/playground/task-scheduler?utm_source=github&utm_medium=social&utm_campaign=TaskScheduler)

If you are interested, I'd love to send a PR and add a link to the library's README. Also, if you have any feedback, I'd love to hear it!

Thanks again for creating and maintaining the awesome library!

arkhipenko, 2020-03-13:
Hi Uri,

I love it! 
How can I contribute more code to it?
Please feel free to send out PRs, link to anything and reference away!
Thank you for doing this.

arkhipenko, 2020-03-13:
This think this section of Wiki is particularly well suited for the site:

https://github.com/arkhipenko/TaskScheduler/wiki/Implementation-scenarios-and-ideas

urish, 2020-03-14:
Great to hear your feedback! 

The easiest way for contributing code right now is simply creating a gist with the code, and sharing the link with me (or you can also share the code through [https://wokwi.com/share](https://wokwi.com/share?utm_source=github&utm_medium=social&utm_campaign=TaskScheduler)).

What code would you like to contribute?

arkhipenko, 2020-03-17:
Just created https://wokwi.com/share/BFqXEGRGRCpmri63q4Wm
Do you want to add text to it (in the comments) to make a challenge out of it?

urish, 2020-03-17:
Lovely! Making a challenge out of it sounds like a great idea, I will do it and post the link here as soon as it's ready

arkhipenko, 2020-03-17:
Can you give me access to making challenges? I have many ideas!I will put this site in the wiki!

urish, 2020-03-17:
Yes, I can open the repo that holds the challenges (basically, a bunch of Arduino sketches, markdown files and some images organized into folders), so that you could start sending pull requests. 

Would that work for you?

arkhipenko, 2020-03-17:
Yep. You can PM to arkhipenko@hotmail.comThx

urish, 2020-03-17:
Here we go:

1. The new challenge is up at https://wokwi.com/u/task-scheduler/challenge-traffic-light
2. I created an open repo for this at https://github.com/wokwi/wokwi-TaskScheduler

Can you please have a look at the repo and let me know if the way things are organized there makes sense to you?

arkhipenko, 2020-03-18:
I don't understand how the repo files relate to the challenge? How do I create LEDs and borders?
Also, will this be accessible from the main landing page? It is not now.
Thanks
Anatoli

urish, 2020-03-18:
Hi Antonio!

My bad, I didn't explain. So here we go:

The sketch source code is [here](https://github.com/wokwi/wokwi-TaskScheduler/blob/master/challenge-traffic-light/sketch.ino), basically just your code.

Then, the explanations and all the hardware can be found inside the [lesson.md](https://github.com/wokwi/wokwi-TaskScheduler/blob/master/challenge-traffic-light/lesson.md) file. It's just a plain markdown file with some HTML code to draw the hardware. More specifically, if you click "Raw" to view the source code on github, you will notice the following code snippet which defines the LEDs and the border around them:

```html
<div style="border:solid black 4px; display: flex; flex-direction: column; background:#ccc; padding:16px 16px; width: 84px">
  <wokwi-led color="red" label="8"></wokwi-led>
  <wokwi-led color="yellow" label="11"></wokwi-led>
  <wokwi-led color="green" label="9"></wokwi-led>
</div>
```

The `label` attribute for each of the LEDs also defines the Arduino pin their anode is connected to (at the moment, the anode is always connected to some pin, either analog or digital, and the cathode to GND). 

There are also other components, such as a pushbutton, I2C LCD display, 7-Segment display and neopixels, and you can check out [the full catalog](https://elements.wokwi.com), or have a look at the [github repo](https://github.com/wokwi/wokwi-elements). They were implemented using type-script and lit-element, and if you are curious, I shared in details [how I created the Arduino pushbutton element](https://www.smashingmagazine.com/2020/01/recreating-arduino-pushbutton-svg/).

If you want to create a new challenge/playground, all you need to do is to create a new directory inside that repo, copy `sketch.ino`, `lesson.md`, and `lesson.json` (which is currently just a placeholder file with an empty object), and change the code in `sketch.ino` and the hardware definition / instructions in `lesson.md`. Does that make sense?

As for the main landing page, yes, it's a big mess right now. It will take a few days until I get to re-organize it, but until then, I have added a section for TaskScheduler with all the resources (both the playground and the red traffic light challenge). Thanks for bringing this up!

arkhipenko, 2020-03-18:
Let's move this conversation to private messages (arkhipenko@hotmail.com) if you don't mind. I will close the issue.

closed 2020-03-18 by arkhipenko

urish, 2020-03-18:
Sure, I will send you an email