arkhipenko/TaskScheduler #91: VisualMicro problem

RomanSmajla, 2020-03-16:
Hi,

I really like your library and using a lot. I'm using Visual Studio with VisualMicro extension. Now with new version i have compiler error. With version Visual.Micro.Arduino.Studio_1912_28_5 everything works fine. With new version Visual Micro - Release 20.03.1 - 16th March 2020 I can't compile code. 
I tray with Nano, Uno, ESP32 and always error. Can you check it please.

Best regards,

arkhipenko, 2020-03-16:
Unfortunately, I do not use or even have Visual Studio with VisualMicro extension installed.
Can you please send the details of the error message. 
Suspect it is something is the settings/paths rather than the library.

RomanSmajla, 2020-03-16:
Hi,

I try to compile your last example "Scheduler_example01" and I get this errors:

Compiling 'bezveze' for 'Arduino Uno'
 
bezveze.ino: In function void t3Callback()
 
bezveze.ino: 56:20: error: 'milli' was not declared in this scope
   Serial.println(millis())
   ^~~~~
Error compiling project sources
bezveze.ino:56: note  suggested alternative  millis
   Serial.println(millis())
   ^~~~~
   millis
Build failed for project 'bezveze'

But then I try to compile next example "Scheduler_example02" and It works. :-)

On next example "Scheduler_example03" get this error

Compiling 'bezveze' for 'Arduino Uno'
 
bezveze.ino: 71:1: error: 'ime' does not name a type
Error compiling project sources
Build failed for project 'bezveze'

very strange things is happening. Maybe is something wrong in Visual Micro extension. I don't know.

Best regards

arkhipenko, 2020-03-16:
Can you compile any sketch that uses millis() without TaskScheduler? 
millis is such a fundamental part of Arduino it feels that the upgrade broke the paths to header files somehow and your Arduino.h is no longer included...

RomanSmajla, 2020-03-16:
Yes I try.

 Serial.begin(115200);
    Serial.println(millis());

compiling OK. I think that is in Visual Micro extension problem because some project compile O.K. with TaskScheduler and some No. I will try to find what is wrong. If I not find problem, I will use older version of exstension and that is it :-)

Thx anyway. Keep updating your library 👍 

Best regards

closed 2020-03-16 by RomanSmajla