arkhipenko/TaskScheduler #92: Simple example for binding member functions on arduino

sticilface, 2020-03-24:
I'm quite new to this lib but i've had a good look around but can't find a solution. 
I'm using AVR so no std libs. 

I'd like to bind member functions to a task.  

I attempted to use the new '_TASK_OO_CALLBACKS` but this appears to then remove the standard constructors for tasks which i need to use as well.  the scheduler is in the main sketch and running the standard example (blink every second), but I need a lib I've written to be able to bind tasks, if i globally define '_TASK_OO_CALLBACKS` then everything stops working, if i include '_TASK_OO_CALLBACKS` prior to `#include <TaskSchedulerDeclarations.h>` in my lib then i get a lot of duplicate errors suggesting that this is not the way to do it. 

I actually wrote my own taster for esp8266 https://github.com/sticilface/Tasker but it is dependent on std... so no use to me. 

many thanks

arkhipenko, 2020-03-24:
You can either use static or dynamic binding - not both at the same time. 
Please use example 21: https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example21_OO_Callbacks
as a template for _TASK_OO_CALLBACKS 

You can also send me your code to arkhipenko@hotmail.com if you want - I'll have a look.

closed 2020-03-25 by arkhipenko

sticilface, 2020-03-26:
Thank you, I am having a deep dive into you lib to understand it more as it is very extensive.  I shall email you if I have any specific questions. thank you.

arkhipenko, 2020-03-26:
Look at Wokwi - a great resource to play with TS code and see an immediate impact of your changes. 

https://wokwi.com/playground/task-scheduler
https://wokwi.com/u/task-scheduler/challenge-traffic-light

sticilface, 2020-03-28:
This is what I ended up with... 
```cpp
template <typename T>
class ooTask : public Task {
public:
  ooTask(uint32_t period, uint32_t duration, T * t, void (T::*fn)() , Scheduler * aS, bool enabled) : Task(period,duration, aS, enabled), _t(t), _fn(fn) {};
  // return true if run was "productive - this will disable sleep on the idle run for next pass ?? meaning of this
  bool Callback() {
    if (_t && _fn) {
      (_t->*_fn)();
    }
  }
private:
  T * _t;
  void (T::*_fn)(); 
};
```

allows you to bind a class member function easily..  need to have `_TASK_OO_CALLBACKS` enabled

arkhipenko, 2020-03-28:
I am discovering new ways of using my own library.  :)))

sticilface, 2020-03-29:
I've sent you an email with example about some trouble i'm having using the `_TASK_OO_CALLBACKS` in `.h` and `.cpp` files.  I'm not sure what is happening, could you take a look?

vortigont, 2020-05-12:
> This is what I ended up with...
> allows you to bind a class member function easily.. need to have `_TASK_OO_CALLBACKS` enabled

A small example on how to use this would be really appreciated :)