arkhipenko/TaskScheduler #95: Run task daily at a specified time

alve89, 2020-04-14:
Is it possible with your library to perform a task lets say every day at 8 PM / 20:00?

arkhipenko, 2020-04-14:
At the core, TaskScheduler is not bound to a specific absolute time. Most microcontrollers are not capable of knowing the absolute time anyway. What you can do it either use Real Time Clock (either hardware or millis()-based, but the latter you need to set every time the mc boots up), or, for WiFi MC's like esp8266, you can run an NTP query, set the time, and then launch at a specific time (which you need to do periodically checking "is it time yet - no, is it time yet, no, is time yet - YES!"  :)

For RTC you could check out [this](https://github.com/arkhipenko/apis/blob/master/apis.ino#L975) repo.
And for NTP [this](https://github.com/arkhipenko/IoT_apis2/blob/master/IoT_apis2.ino#L948) one. 

Both are a bit large and complex, but I am sure you can isolate the needed components and reuse.

arkhipenko, 2020-04-14:
Interesting idea for a future feature option though.

arkhipenko, 2020-04-14:
you can also set alarm on an RTC, and trigger a task via pin interrupt.

alve89, 2020-04-14:
Thanks for your quick reply!

I'm using an ESP32 and wanted to run a task daily. Since I read the ESP32's RTC is not the best I'm trying to use my DS1307 RTC module. So by now I'm used to NTP and RTC.
BUT: By receiving the current timestamp via NTP I'm also getting the date (!) and time. And the date is my problem, because I don't know how to set a specified datetime as absolute begin (like 00:00:00). From there I could calculate 24h x 3600s/h. And here I thought I could your library...

arkhipenko, 2020-04-14:
I believe the library I used had an option of using Unix time, which is number of seconds from some date in 1960... Easy to work with when you don't need more than a second precision.

alve89, 2020-04-14:
But not directly with your library and it’s functions of „multitasking“, right?

arkhipenko, 2020-04-14:
No. TS currently does not support scheduling based on specific time/date.

alve89, 2020-04-14:
Very pity, but thanks for your quick help.

closed 2020-04-14 by alve89