arkhipenko/TaskScheduler #96: Problem with OnEnable parameter of task definition

intabits, 2020-04-21:
Hi, 
I've just joined GitHub, just to post this. Sorry if this not the correct way to ask/report this issue. 

In Arduino, my OnEnable handler is not being found. (first time I've tried it)
I was typing in an example here to show you when I saw the problem - I needed a forward definition for my OnEnable handler.

But the example 5 in the implementation scenarios fails in the same way as mine was.
So now this is just to let you know that.

arkhipenko, 2020-04-22:
I need more information and possibly the code. I can't help you based on what you sent.
What is the chip you are compiling on?  What are the compile options you are using? 
A code sample will be the best.

arkhipenko, 2020-04-22:
> But the example 5 in the implementation scenarios fails in the same way as mine was. 

Example 5 was written before Arduino IDE started to require method forward definitions. It used to compile just fine. Regardless, I updated it and it compiles now.

closed 2020-04-24 by arkhipenko