arkhipenko/TaskScheduler #100: Include example for one-off task

skorokithakis, 2020-07-12:
I have a relay I need to turn on and then off after N seconds. I see the LED example, but it defines multiple tasks, and understanding the flow becomes a bit complicated. Is there an example where a single, one-off task is scheduled for N seconds, with an "on" and an "off" handler? That would be very much appreciated.

arkhipenko, 2020-07-12:
Easy. Define a task:Scheduler ts;Task tRelay (relayOnTime * TASK_SECOND, 2, &callback, &ts);void callback () { if ( tRelay.isFirstIteration() ) {  turnRelayOn();  return;}turnRelayOff();}Initiate the task with this:tRelay.enableDelayed( initial delay );That's it. Hope this helps.

skorokithakis, 2020-07-12:
Thank you! Also, is there a way to cancel the task that's already scheduled? Sometimes I have to turn the relay off in an emergency and would like the task (or all tasks) to cancel at that point.

arkhipenko, 2020-07-12:
tRelay.disable().There is a Wiki, you know, that you need to read :)

skorokithakis, 2020-07-12:
I searched for "cancel" and "stop" and didn't find this, sorry 😟

arkhipenko, 2020-07-12:
https://github.com/arkhipenko/TaskScheduler/wiki/API-Task

closed 2020-07-12 by arkhipenko